ARM-DOE/pyart #932: No module named  "pyart.io.sband_archive"?

huyaox, 2020-05-18:


huyaox, 2020-05-18:
No module named "pyart.io.sband_archive"?

zssherman, 2020-05-18:
@huyaox What is the origin of this file, did you try the io.read auto read function? There isn't a sband_archive function in Py-ART.

huyaox, 2020-05-18:
after installing pyart,when run thin "from pyart.io.sband_archive import read_sband_archive",then show
No module named "pyart.io.sband_archive"，how to solve？

huyaox, 2020-05-20:
after instaling pyart ,when run this "from pyart.io.sband_archive import read_sband_archive", show the results"No module named "pyart.io.sband_archive" how to resolve it?

zssherman, 2020-05-20:
@huyaox Have you tried using just pyart.io.read? There is not a pyart.io.sband_archive function in pyart which is what is throwing the error.

huyaox, 2020-05-21:
there is article https://cloud.tencent.com/developer/article/1471399  ，said “from pyart.io.sband_archive”

import matplotlib.pyplot as plt

import pyart
from pyart.io.sband_archive import read_sband_archive

filename ='Z_RADR_I_Z9250_20190409022800_O_DOR_SA_CAP.bin'
sradar =read_sband_archive(filename)

fig, ax = plt.subplots(figsize=(12,9))

srdisplay = pyart.graph.radardisplay.RadarDisplay(sradar)
srdisplay.plot_ppi('reflectivity',0, vmin=0, vmax=70)

fig.savefig('sa_radar.png', dpi=300, bbox_inches='tight')

zssherman, 2020-05-21:
Ah I see! @huyaox So it seems the authors of the paper they have a forked version of pyart and added to it new readers for the c98 and sband radars, according to the article. I'm not sure what user has the forked though from reading the article. You might have to contact the authors and see if their code is shareable, or if you can clone their fork. @scollis thoughts?

scollis, 2020-05-21:
Same @zssherman . We can not provide support for a feature in a fork we do not maintain.

lingyunfeng, 2020-08-06:
I know this forked version of pyart,  which is forked from https://github.com/bugsuse/pyart/tree/master/pyart/ . read_sband_archive functon is for  chinese sband radars,  I hope read_sband_archive functon  can be merged  to master version of pyart.    @zssherman  @huyaox

zssherman, 2020-08-07:
@lingyunfeng Yeah, I think it would be a great addition. If bugsuse is the contributor for that would have to see if they would want to do a PR and depending on unit tests etc add it to the aux_io or io.

niguanne-coder, 2021-04-10:
cannot import name '_fast_interpolate_scan_4' from 'pyart.io.nexrad_interpolate' (C:\Users\ghd\anaconda3\lib\site-packages\pyart\io\nexrad_interpolate.cp38-win_amd64.pyd)?

zssherman, 2021-10-11:
Will leave open, if others come across this