ARM-DOE/pyart #933: Support for RPG Build 19.0 Level III Nexrad Files

dcedgren, 2020-05-20:
Issue #887 documented upcoming changes to the RPG with Build 19.0. However, the issue and its PR focused on changes to Level II file formats. There are also a couple upcoming changes to Level III files. Namely:

- Product 171 is being removed. No change will be required in Pyart and it will continue to support old product 171 files.
- The draft ICD documents that the version number for Product 172 will be incremented from Version 1 to Version 2. It gives this explanation: "Version 2 added one new parameter to the Supplemental Data for the KDP Multiplier for Rain/Hail and three new parameters for the Specific Attenuation Rain Rate". I was not able to find a specific reference to document these four new parameters, and I do not believe reading these parameters is something Pyart supports. Pyart also does not support reading tabular alphanumeric data from Level III files, so no change should be required here.
- Product 34 is being removed. No change will be required in Pyart and it will continue to support old product 34 files.
- Product 131 is being added.

I tested reading a beta file for Product 172 from the Dodge City radar and it failed. The file claimed to still be version 1 instead of version 2, the packet code was listed as 1, rather than 16 like I would expect, and the compression was different than the original file type. It also didn't appear to read in right when I forced it to read in in radial format while debugging. So either the ROC still has a couple changes to make or Product 172 files might have changed more than I thought. I suggest leaving this issue open until July when the operational version can be tested. 

I tested reading a beta file for Product 113 from the Dodge City radar. Due to above I'm not 100% sure this will be the final format. However, the Packet Code was listed as -20705. When the new product becomes operational I suggest testing again and adding to the list of unsupported radial format products if required.

A good summary of changes to the RPG in Build 19.0 is located here: [https://www.weather.gov/media/notification/scn19-97wsr-88d_build19.pdf]

Here's a copy:
> NOUS41 KWBC 221450
> PNSWSH
> Service Change Notice 19-97
> National Weather Service Headquarters, Silver Spring, MD
> 1050 AM EST Fri Nov 22 2019
> To: Subscribers
>  -NOAA Weather Wire Service
>  -Emergency Managers Weather Information Network
>  -NOAAPORT
>  Other NWS Partners and Employees
> From: Terrance J. Clark
>  Director, WSR-88D Radar Operations Center
> Subject: RDA/RPG Build 19.0 to begin Beta Test on or around
>  April 15, 2020
> 
> The NWS Radar Operations Center will begin a Beta test of Radar
> Data Acquisition (RDA)/Radar Product Generator (RPG) Build 19.0
> on or around April 15, 2020. Beta test sites will be determined
> at a later date. Full deployment is expected on or around June
> 30, 2020.
> Changes pertinent to external users are as follows:
>
> 1) Baseline of Supplemental Low Elevation Angles
> KMAX -0.2 Medford, OR
> KGJT -0.2 Grand Junction, CO
> KFSX -0.2 Flagstaff, AZ
> KMSX -0.2 Missoula, MT
> KMTX 0.0 Salt Lake City, UT
> KRGX 0.0 Reno, NV
> KDLH +0.2 Duluth, MN
> KCLX +0.3 Charleston, SC
> KCLE +0.4 Cleveland, OH
>
> 2) Product 171/STA Dual Pol Storm Total Accumulation (16 data
> level) will be removed. The alphanumeric data from this product
> will be moved to Product 172/DSA Dual Pol Digital Storm Total
> Accumulation (256 data level).
> The Public Information Statement proposing this change was
> issued in December 2018:
> https://www.weather.gov/media/notification/pdfs/pns18-
> 35wsr_88d_p171termination.pdf
> No public comments objecting to the change were received.
>
> 3) Volume Coverage Pattern (VCP) 121 will be removed and
> replaced with VCP 112. Details of VCP 112 are available:
> https://www.roc.noaa.gov/WSR88D/NewRadarTechnology/NewTechDefault.aspx
>
> 4) Addition of a new field to the Archive Level II radial data 
> stream containing the amount of clutter power removed. In
> association with this change, Product 34/CFC Clutter Filter
> Control will be removed and replaced with elevation-based
> Product 113/PRC Power Removed Control. More details are
> available in the draft Interface Control Documents.
>
> 5) The Level II Differential Reflectivity (ZDR) scale will be
> expanded to a range of -13 dB to +20 dB.
>
> Draft RDA/RPG Build 19.0 Interface Control Documents (ICDs) are
> available:
> https://www.roc.noaa.gov/WSR88D/NewRadarTechnology/NewTechDefault.aspx
>
> Please direct questions to:
> Jessica Schultz
> Deputy Director
> Radar Operations Center
> Jessica.A.Schultz@noaa.gov
> National Public Information Statements are online at:
> https://www.weather.gov/notification/
> NNNN

zssherman, 2020-05-21:
Thanks @dcedgren, I'll look into this! And yeah, we can leave this open until July and test everything out.

dcedgren, 2020-07-30:
Status update. I went today and tried to find the new Product 131 today for testing. I wasn't able to find it. I don't run in the right circles to know, but perhaps deployment was delayed? This issue probably deserves revisiting again in another few months for eventual testing.

dopplershift, 2020-08-18:
Looks like things are live, both the beta sites as well as the sites running the low elevation angles. It turns out the product 113, being basically a clutter map, isn't available on NOAAPORT. You can get them from the NWS TG, or we (Unidata) seem to have them (somehow) hiding in the LDM HDS feed (under WMO header SDUS6). They have the product code N[0123ABYX]F (like N0Q).

The product 172 I pulled looked fine and read into MetPy fine. 113 was a bit more tricky. The 2620001Y ICD doesn't say anything about 113 having the product-specific bytes indicating compression, but:
1. Those bytes definitely have data
2. The only way the product decodes without error for me is by setting up the usual stuff for compression in newer radial products.
Thought I'd share my experience from putting this into MetPy.

I went ahead and attached a zip of some of the [new products](https://github.com/ARM-DOE/pyart/files/5088414/New.NIDS.zip) (as well as the modified DTA 172)--all of 500kb.