ARM-DOE/pyart #935: Can't remove colorbar from graph

LucasGood, 2020-06-02:
```
display=pyart.graph.RadarDisplay(radar)
        if product=='reflectivity':
            vmin=0
            vmax=94
        if product=='differential_reflectivity':
            vmin=0
            vmax=8
        if product=='velocity':
            vmin=-142
            vmax=142
        if product=='cross_correlation_ratio':
            vmin=0.208
            vmax=1.052
        display.plot(product, tilt, title='', vmin=vmin, vmax=vmax, colorbar_flag=False, title_flag=False, axislabels_flag=False)
        fig=plt.figure(figsize=[30,30]) # figsize determines the resolution of the rendering, higher numbers = higher load times
        fig.subplots_adjust(left=0, bottom=0, right=1, top=1, wspace=0, hspace=0)
        display.plot(product, tilt, title='', vmin=vmin, vmax=vmax)
        display_limit=-230, 230
        display.set_limits(xlim=(display_limit), ylim=(display_limit))
        plt.xlim(-235,235)
        plt.ylim(-235,235)
        F=plt.gcf()
        Size=F.get_size_inches()
        F.set_size_inches(Size[0]*1.5, Size[1]*1.5, forward=True)
        plt.gca().set_aspect('equal', adjustable='box')
        plt.savefig(site + ".png", transparent=True)
```
I need to make an image that graphs only the radar image and nothing else.
Although  `colorbar_flag=False`, the saved image still has a color bar on the right. 

[image]

I've been looking through examples and the documentation and I haven't found a definitive way to remove this colorbar. 

Let me know if I need to look elsewhere or provide more information.

zssherman, 2020-06-02:
Hi @LucasGood I don't see a colorbar_flag in the second display.plot so I think the second plot is overwriting the second one, maybe try adding the colorbar_flag=False to your second display.plot

LucasGood, 2020-06-02:
That fixed it!

Thanks for being patient, sir.

closed 2020-06-02 by LucasGood

zssherman, 2020-06-02:
No problem and glad to hear it!