ARM-DOE/pyart #936: Issue with multiple-sweep RHIs with NPOL Sigmet data

wxwolff, 2020-06-02:
Using NPOL Sigmet RHI data with 3-sweeps: 195, 197 and 199 degrees.  First sweep is fine with data only present in ranges > 0 (NPOL can only scan 0-90 deg vertically). Unexpectedly the middle sweep (sweep 1) has data in both positive and negative ranges. The final sweep (sweep 2) is, like sweep 0, ok.  I have tested this with new and old data and with code I copied from the Pyart Plotting Examples and get the same result.

Suggestions?  I can make a sample file available for testing if anyone wants to help.

[image]
[image]
[image]

scollis, 2020-06-02:
Thanks Dave! Yes a file would be great. How was Py-ART installed? Can you post pyart.__version__ 

-sent from a mobile device-

wxwolff, 2020-06-03:
**I am using Anaconda (on mac 10.14 and linux Centos 6.10 (Final)).** 

conda info
     active environment : None
            shell level : 0
       user config file : /Users/wolff/.condarc
 populated config files : 
          conda version : 4.8.3
    conda-build version : 3.18.11
         python version : 3.7.7.final.0
       virtual packages : __osx=10.14.6
       base environment : /Users/wolff/anaconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/wolff/anaconda3/pkgs
                          /Users/wolff/.conda/pkgs
       envs directories : /Users/wolff/anaconda3/envs
                          /Users/wolff/.conda/envs
               platform : osx-64
             user-agent : conda/4.8.3 requests/2.23.0 CPython/3.7.7 Darwin/18.7.0 OSX/10.14.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False

>>> import pyart
## You are using the Python ARM Radar Toolkit (Py-ART), an open source
## library for working with weather radar data. Py-ART is partly
## supported by the U.S. Department of Energy as part of the Atmospheric
## Radiation Measurement (ARM) Climate Research Facility, an Office of
## Science user facility.
##
## If you use this software to prepare a publication, please cite:
##
##     JJ Helmus and SM Collis, JORS 2016, doi: 10.5334/jors.119
/Users/wolff/anaconda3/lib/python3.7/site-packages/osgeo/__init__.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
>>> pyart.__version__
**'1.11.1'**

Attached is a sample RHI file...

[np1200430153119.RAWAJG1.gz](https://github.com/ARM-DOE/pyart/files/4722702/np1200430153119.RAWAJG1.gz)

scollis, 2020-06-03:
@zssherman can you take a look

zssherman, 2020-06-03:
@wxwolff Interesting, yep @scollis I will take a look.

zssherman, 2020-06-03:
Oh wait you do have the most recent pyart...

wxwolff, 2020-06-03:
I am running 1.11.1 using Anaconda3.  

I grabbed some code from the PyART Examples to verify this wasn't a coding issue on my part. Note that the issue became apparent to me when I did not set xlim. In the code below you can set sweep = 0, 1 or 2.

Thanks!

import matplotlib.pyplot as plt
import pyart

filename = '/Users/wolff/Desktop/Data/NPOL/RHI/np1200430153119.RAWAJG1.gz'

# create the plot using RadarDisplay
radar = pyart.io.read_rsl(filename)
display = pyart.graph.RadarDisplay(radar)

fig = plt.figure(figsize=[10, 4])
ax = fig.add_subplot(111)

# instrument_name = radar.metadata['instrument_name'].decode('utf-8')
# time_start = netCDF4.num2date(radar.time['data'][0], radar.time['units'])
# time_text = ' ' + time_start.strftime('%Y-%m-%dT%H:%M:%SZ')
azimuth = radar.fixed_angle['data'][0]
title = filename

sweep = 1
display.plot('reflectivity', sweep, vmin=0, vmax=65,
             title=title, colorbar_flag=False, ax=ax)
display.set_limits(ylim=[0, 12])

cax = fig.add_axes([.9, .1, 0.02, .8])
colorbar_label = 'Eq refl fact (dBz)'
display.plot_colorbar(fig=fig, cax=cax, label=colorbar_label)

plt.show()

zssherman, 2020-06-03:
Ah ok, @wxwolff @scollis so i narrowed it down, but still figuring out why. So your using read rsl, when using that it produces the negative ranges. When I used auto read in my code, it uses read sigmet and the negative ranges don't appear. The different im not sure yet, but one of the readers is doing something to remove or ignore negative ranges or elevations, so i'll need to check where.

zssherman, 2020-06-03:
Elevation when reading with sigmet
```
print(radar.elevation['data'].max())
print(radar.elevation['data'].min())
25.081785
-0.02472378
```
With rsl:
```
359.97528
0.008239746
```

wxwolff, 2020-06-03:
@zssherman   @scollis   This is definitely a SIGMET file. So I changed to the auto reader and no longer have the issue.  I assumed it was rsl because NPOL was one of the first radars we used in developing RSL.   Many thanks for your help!!

closed 2020-06-03 by wxwolff