ARM-DOE/pyart #940: gatefilter not working in `pyart.map.grid_from_radars`

swnesbitt, 2020-07-01:
I'm not having any luck getting a gatefilter to remove bad gates when passed to `pyart.map.grid_from_radars`.  I spoke to @rcjackson on Slack and he suggested using the `gatefilter.gates_excluded` to mask the data, but this does not seem to be functioning as intended.

[image]

scollis, 2020-07-01:
Can you try something crazy like excluding Z> 20

What did you construct the filter from?

-sent from a mobile device-

swnesbitt, 2020-07-01:
# filter out gates with reflectivity > 100 from both radars
gatefilter_1 = pyart.filters.GateFilter(radar1)
gatefilter_1.exclude_transition()
gatefilter_1.exclude_below('normalized_coherent_power', 0.3)
gatefilter_1.exclude_above('differential_reflectivity', 5)
pyart.correct.despeckle.despeckle_field(radar1, 'reflectivity', label_dict=None, threshold=-100, size=10, gatefilter=gatefilter_1, delta=5.0)
gatefilter_1.exclude_masked('reflectivity')
gatefilter_1.exclude_masked('velocity')

swnesbitt, 2020-07-01:
I tried excluding reflectivity > 20 - works in the plot, but not the grid again.

[image]

swnesbitt, 2020-07-01:
Was able to workaround with
```
radar1.fields['reflectivity']['data']=np.ma.masked_where(gatefilter_1.gate_excluded,radar1.fields['reflectivity']['data'])
radar1.fields['velocity']['data']=np.ma.masked_where(gatefilter_1.gate_excluded,radar1.fields['velocity']['data'])
```
Off to `pydda` land!

scollis, 2020-07-01:
Cool beans. Zach can you use a gate filter on some data to see if it works on our end 

-sent from a mobile device-

zssherman, 2020-07-01:
Yeah, I'll take a look and see if I can figure out what's going on.

zssherman, 2020-07-02:
Yeah, @swnesbitt @scollis 
I'm able to reproduce, was just trying to exclude below reflectivity 0 and still shows up. I'll have to look into whether or not there was a numpy update that changed mask behavior or something. 
[image]

zssherman, 2020-07-17:
@swnesbitt I think it might be a typo, can you have gatefilter parameter be called gatefilters, gatefilters is a kwarg parameter that gets plugged into map_gates_to_grid, and if its gatefilter, it will pass an unknown parameter to kwargs and pyart then will set gatefilter to a basic moment gatefilter thinking gatefilters is False. I tried gatefilter and got no mask, provided gatefilters and got the mask.

zssherman, 2021-10-11:
Closing for now

closed 2021-10-11 by zssherman

msteckle, 2022-12-29:
Hello, I see this is closed, but I am using grid_from_radars and am trying to display gate filtered images. But the gate filter doesn't seem to be working (unless I'm missing something obvious). Below is my code (running on a Jupyter notebook) and results.

``` python
def map_radar(radar):
    
    # remove ring around radar
    radar.fields['reflectivity']['data'][:, -10:] = np.ma.masked

    # specify which gates to exclude
    gatefilter = pyart.filters.GateFilter(radar) # instantiate gate filer
    gatefilter.exclude_transition() # ignore gates collected between sweeps
    gatefilter.exclude_masked('reflectivity') # ring mask we created
    gatefilter.exclude_below('reflectivity', 0) # ignore values below 0
    
    # create grid
    grid_obj = pyart.map.grid_from_radars(radars=(radar),
                                          gatefilters=(gatefilter),
                                          grid_shape=(1,
                                             int(radar.range['meters_between_gates']),
                                             int(radar.range['meters_between_gates'])),
                                          grid_limits=((radar.gate_z['data'].min(), 
                                               radar.gate_z['data'].max()),
                                              (radar.gate_x['data'].min(),
                                               radar.gate_x['data'].max()),
                                              (radar.gate_y['data'].min(),
                                               radar.gate_y['data'].max())))
    
    # display grid object
    ax = plt.subplot()
    im = ax.imshow(grid.fields['reflectivity']['data'][0], 
              origin='lower',
              interpolation=None,
              cmap='pyart_NWSRef')
    divider = make_axes_locatable(ax)
    cax = divider.append_axes("right", size="5%", pad=0.05)
    plt.colorbar(im, cax=cax)
    plt.show()

map_radar(radar_object)
```

[image]

You can see that values below 0 and ring are still being displayed. I haven't tried the workaround mentioned above yet, but I wanted to point this out anyway. Thanks!