ARM-DOE/pyart #941: 180 Degree PPI from https://arm-doe.github.io/pyart/API/generated/pyart.util.cross_section_ppi.html

rrniii, 2020-07-02:
Would it be possible to create an option to create a full 180 degree RHI (i.e. on that crosses zenith) from a volume of PPI data?

zssherman, 2020-07-02:
I'll look into this, there actually might be a bug in the cross section code. I was taking a look now  and specified 3 desired cross sections and received 51 sweeps... But I'll try to get the correct behavior going and then take a look at this enhancement

rrniii, 2020-07-02:
Thanks Zach. A few members of my group would find it useful. The title of this request should have been “180 degree RHI from....” sorry about that. Not enough brain power left over here at the end of my day in Leeds.

closed 2022-06-10 by mgrover1