ARM-DOE/pyart #942: map_gates_to_grid creates blobish edges when weight function set to nearest

cptmedic, 2020-07-07:
Good evening,

Zach asked that I create a Github Issue for the following issue. 

Previously when generating a geotiff, map_gates_to_grid would generate a radar file with blue aliasing throughout (pictured below):
[image]

Merge Request #934 fixed the blue aliasing, but seemed to introduce a new issue where edges of storms would be blobish, and filtering did not remove it.
[image]

Changing the algorithm to map_to_grid produces an accurate reflection of the storm, but takes a significantly longer time to generate:
[image]

It appears that there is an issue somewhere with the map_gates_to_grid cython code that is causing the distortion at the boundaries to radar returns.

zssherman, 2020-07-07:
Thanks @cptmedic! @rcjackson this is in the nearest neighbor cython code whether you get a chance probably after the conference this week if we can work together and sprint to fix this?

zssherman, 2020-07-15:
@cptmedic This issue might have been found, can you share this file or how are you obtaining the file in the top two images?

zssherman, 2020-07-15:
To test it out

cptmedic, 2020-07-17:
@zssherman I have emailed you the python script in it's entirety

zssherman, 2020-07-22:
@cptmedic In #947 this should now be fix! I have to push out a release still, but in the mean time you can get the fix by git cloning and installing the pyart repo itself.

cptmedic, 2020-07-24:
Installed from repo source and can confirm it works great! Thanks guys!

closed 2020-07-24 by cptmedic