ARM-DOE/pyart #943: Unicode error when loading Py-ART Grids: add explicit conversion to Unicode?

rcjackson, 2020-07-13:
A reviewer of my PyDDA attempted to load .nc Grid files I had created with PyART and ran into an unusual error. I suspect his system is using ASCII as the default encoding for strings, but when I look at netCDF4's issue tracker, they use utf8 for their encoding. Therefore, when he tried to use pyart.io.read_grid, the reviewer gets a UnicodeDecode error. One workaround for any user for now would be to explicitly specify utf8 encoding by adding a u' before their string, but I wonder if we can check for string encoding before we load the grid file and convert to utf8 if needed.

Has anyone ever ran into this error? This grid loads fine on any system I have ever used, so I suspect it's the reviewer's default encoding for his system.

run hurricane_florence.py                                                                                
---------------------------------------------------------------------------
UnicodeDecodeError                        Traceback (most recent call last)
/afs/ies.auc.dk/user/tha/Reviews/JORS - Jackson 2020/hurricane_florence.py in <module>
 	26 urllib.request.urlretrieve(hrrr_url, 'test.grib2')
 	27
---> 28 grid_mhx = pyart.io.read_grid('grid_mhx.nc')
 	29 grid_ltx = pyart.io.read_grid('grid_ltx.nc')
 	30

~/.conda/envs/pydda/lib/python3.7/site-packages/pyart/io/grid_io.py in read_grid(filename, exclude_fields, include_fields, **kwargs)
 	57         'ProjectionCoordinateSystem']
 	58
---> 59 	dset = netCDF4.Dataset(filename, mode='r')
 	60
 	61 	# metadata

netCDF4/_netCDF4.pyx in netCDF4._netCDF4.Dataset.__init__()

netCDF4/_netCDF4.pyx in netCDF4._netCDF4._ensure_nc_success()

UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 7: ordinal not in range(128)

zssherman, 2020-07-13:
@rcjackson I personally have never run into this before, but I do agree having a checker to convert to utf8 would be a good route to go.

rcjackson, 2020-07-13:
Add that to the list of things to do on our Wednesday sprint.

zssherman, 2020-07-13:
Haha yep, sounds good to me

closed 2022-11-03 by zssherman