ARM-DOE/pyart #951: cannot install pyart as part of requirements.txt because of missing numpy

kracwarlock, 2020-07-28:
https://github.com/ARM-DOE/pyart/blob/3bbab4a012349559874e0923b5a5c4f1962d6540/setup.py#L165
`pyart` imports `numpy` in `setup.py` so if you are setting up `arm_pyart` within a `requirements.txt` file it will fail since `numpy` is not pre-installed. it would be better if numpy could be listed as a dependency instead and python's distutils used to setup the package.

zssherman, 2020-07-30:
@kracwarlock Yeah, I agree, we do plan on moving the setup.py to setuptools in the future. Just sadly haven't gotten to it yet.

Vforcell, 2022-02-23:
@kracwarlock @zssherman The same problem is persisting for me. Do you have any update on that issue ?

zssherman, 2022-03-14:
Closing in favor of #823

closed 2022-03-14 by zssherman