ARM-DOE/pyart #952: Read Rainbow 5.37.6 file with different number of bins between sweeps

pedroaugustosmribeiro, 2020-08-03:
Please check if it's possible to parse a RAINBOW file version 5.37.6 that gives the error: "Number of ranges bins changes between sweeps"
It's readable on wradlib.

zssherman, 2020-08-07:
@pedroaugustosmribeiro Yeah, might need to add a max bin and mask bins added to lower bin sweeps for sweeps with differing shapes. I believe something similar to this exists in one of the nexrad readers, so I'll see if I can track it down.

tsbiscaro, 2023-03-01:
Any update on this issue?

zssherman, 2024-05-09:
Sorry for the very late response. Our goal is eventually to use xradar as the io portion of reading radar files into xarray. I believe:
https://docs.openradarscience.org/projects/xradar/en/stable/notebooks/Rainbow.html
should handle different number of bins between sweeps.
Then can use it within pyart as such:
https://arm-doe.github.io/pyart/examples/xradar/plot_xradar.html#sphx-glr-examples-xradar-plot-xradar-py