ARM-DOE/pyart #953: Support for additional weighting functions in grid mapping (including range- and time-dependence)

jthielen, 2020-08-04:
In the grid mapping routines (`grid_from_radars`, `map_gates_to_grid`, and `map_to_grid`), would there be support for expanding the `weighting_function` options that are available to use, either by additional built-in options or refactoring to support custom functions?

My particular use-case is re-implementing the [GridRad procedure](http://gridrad.org/pdf/GridRad-v3.1-Algorithm-Description.pdf) in `map_gates_to_grid`. It uses nearest neighbor interpolation in the horizontal and a vertical depth cutoff combined with a range and time offset Gaussian weight. While I'm working on a vendored-and-modified implementation for my own needs (see initial work here: https://github.com/jthielen/OpenMosaic/blob/560bc90f8cada41a1817f2261c8002f13cc16eae/src/openmosaic/gridding/_gate_to_grid_map.pyx), I was wondering if such an addition would be welcome upstream in Py-ART, or at least made easier to do through custom weighting function support.

If built-in additions are welcome, I'd imagine some version of the GridRad procedure and/or some combination of 4DDG procedures of [Langston et al. (2007)](https://journals.ametsoc.org/jtech/article/24/5/776/2999/Four-Dimensional-Dynamic-Radar-Mosaic) would be particularly useful to include.

zssherman, 2020-08-04:
I would say having an option for custom weights or more built in weights would be great. @scollis Your thoughts?

scollis, 2020-08-04:
100% yes