ARM-DOE/pyart #954: Building Py-ART with TRMM RSL Support.

stinchy, 2020-08-14:
Hi Guys,   

I have been trying to build Py-ART with TRMM RSL support as is outlined on the website [here.](http://arm-doe.github.io/pyart/INSTALL.html)  
I installed TRMM RSL v1.50 and have verified the locate at /usr/local/trmm.  I added a bash global variable to point $RSL_PATH to /usr/local/trmm.  I then cloned Py-ART through git and then ran python3 setup.py install.   However, this apparently did not build with rsl support as when I run my script with it, I get:
`/usr/lib/python3/dist-packages/osgeo/__init__.py:8: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Traceback (most recent call last):
  File "BLSN_UND_radar.py", line 14, in <module>
    radar = pyart.io.read_rsl(fname)
  File "/usr/local/lib/python3.6/dist-packages/pyart/io/rsl.py", line 85, in read_rsl
    "read_rsl function.")
pyart.exceptions.MissingOptionalDependency: Py-ART must be build with support for TRMM RSL to use the read_rsl function.
`

Anybody have any idea what I might be doing wrong?

zssherman, 2020-08-17:
Hmm, have you tried using conda environments? When it comes to locally installed trmm_rsl I don't have much experience. But typically with conda trmm_rsl is conda installed and the path is set to the environment to avoid issues with base packages etc. Also I wonder if trmm_rsl needs to be in dist-packages similar to where pyart and your other modules are. Wondering if python is not finding anything outside of dist_packages.

stinchy, 2020-08-18:
HI @zssherman.  

I am now trying to use Anaconda3 as is also outlined in the Py-ART install page.  I am trying to install the pyart environment from  https://github.com/ARM-DOE/pyart/blob/master/environment.yml  

When I run `conda env create -f environment.yml`  I get errors that appear to be related to a formatting issue with the yaml file on line 124 which is the following:
`<span style="background-color: #79b8ff;width: 0%;" class="Progress-item progress-pjax-loader-bar "></span>`
The error is pointing at the colon after background-color.  I assume that the colon or some other part of that line is not getting escaped correctly or it is getting read out of context, but I'm not well versed with yaml, so I'm not quite sure what I need to fix the line.

Any Idea?

zssherman, 2020-08-19:
@stinchy Thats odd, that doesn't seem like a yml issue. That looks like a html error, which not sure where that is originating from. Was there other messages with the enviroment creation? I'm honestly not sure about the yml error. You could also trying;
conda create -n pyart_env python=3.7 numpy scipy cartopy netCDF4 trmm_rsl matplotlib
then set path to (an example) export RSL_PATH=/home/zsherman/anaconda3/envs/pyart_env/
just have to point to the environment not trmm_rsl directly