ARM-DOE/pyart #955: dtype not preserved after gridding

vlouf, 2020-08-19:
The dtype of input fields is not preserved by the gridding operation. For example, in the original ppi radar, we have these fields:
```python
{'radar_echo_classification': dtype('int16'),
 'radar_estimated_rain_rate': dtype('float32'),
 'velocity': dtype('float32'),}
```

And after gridding everything is float32. Currently I have this little piece of code to correct the types of the input fields:
```python
intypes = {k: v['data'].dtype for k, v in radar.fields.items()}
for k, v in intypes.items():
    grid.fields[k]['data'] = grid.fields[k]['data'].astype(v)
```
But it would be nicer if the gridding preserved the type (or automatically correct it right after gridding).

scollis, 2020-08-19:
Thanks @vlouf Agree

zssherman, 2020-08-19:
@scollis @vlouf See #956