ARM-DOE/pyart #958: gate_longitude['data'] method for NEXRAD Level 3 no longer working in 1.11.2

guidodev, 2020-09-11:
Normally, the following code...

```
import pyart
import numpy as np

dhr_scan = pyart.io.read_nexrad_level3('dhr_test.nexrad3')
lons = dhr_scan.gate_longitude['data']
lats = dhr_scan.gate_latitude['data']
print(lons)
print(lats)
```
...correctly yields:

```
[[-98.412994 -98.412994 -98.412994 ... -98.412994 -98.412994 -98.412994]
 [-98.412994 -98.412766 -98.412544 ... -98.36028  -98.36005  -98.3598  ]
 [-98.412994 -98.412544 -98.4121   ... -98.30759  -98.30711  -98.306625]
 ...
 [-98.412994 -98.413666 -98.41434  ... -98.57105  -98.57178  -98.5725  ]
 [-98.412994 -98.413445 -98.41389  ... -98.518394 -98.518875 -98.51936 ]
 [-98.412994 -98.413216 -98.413445 ... -98.465706 -98.46594  -98.46619 ]]
[[45.456    45.464993 45.47398  ... 47.49698  47.505962 47.514946]
 [45.456    45.464993 45.47398  ... 47.496655 47.50564  47.51462 ]
 [45.456    45.46499  45.47397  ... 47.495686 47.504665 47.51364 ]
 ...
 [45.456    45.46498  45.473957 ... 47.494072 47.503044 47.512012]
 [45.456    45.46499  45.47397  ... 47.495686 47.504665 47.51364 ]
 [45.456    45.464993 45.47398  ... 47.496655 47.50564  47.51462 ]]
```

With 1.11.2, the result is the following error:

```
Traceback (most recent call last):
  File "./junk.py", line 7, in <module>
    lons = dhr_scan.gate_longitude['data']
  File "/home/testuser5/.local/lib/python3.6/site-packages/pyart/lazydict.py", line 66, in __getitem__
    value = self._lazyload[key]()
  File "/home/testuser5/.local/lib/python3.6/site-packages/pyart/core/radar.py", line 994, in _gate_lon_lat_data
    x = radar.gate_x['data']
  File "/home/testuser5/.local/lib/python3.6/site-packages/pyart/lazydict.py", line 66, in __getitem__
    value = self._lazyload[key]()
  File "/home/testuser5/.local/lib/python3.6/site-packages/pyart/core/radar.py", line 978, in _gate_data
    ranges, azimuths, elevations, edges=False)
  File "/home/testuser5/.local/lib/python3.6/site-packages/pyart/core/transforms.py", line 117, in antenna_vectors_to_cartesian
    return antenna_to_cartesian(rg / 1000., azg, eleg)
  File "/home/testuser5/.local/lib/python3.6/site-packages/pyart/core/transforms.py", line 67, in antenna_to_cartesian
    theta_e = elevations * np.pi / 180.0    # elevation angle in radians.
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'

```
I've tested 1.10.1, 1.10.4, 1.11.0, 1.11.1, and 1.11.2. Only 1.11.2 appears to be affected by this problem, so my workaround for now is sticking with an older version.

Thanks in advance!

P.S. I don't know if this is related, but it seems worth noting that all 1.11.x versions require cftime and scipy to be explicitly specified in requirements.txt. Without them, the above test script crashes on other issues. By contrast, 1.10.x runs the code with arm-pyart as the only package specified in requirements.txt.

zssherman, 2020-09-11:
Thanks @guidodev for reporting! I'll have to take a look and see what could be causing this. Coordinate functions havent been touched, atleast that I'm aware of, so I'm curious what could have changed. And doesnt look like your using conda so, my thought of a dependency issue with the new version is out the window. I'll have to dig deeper into what is causing this, but thanks for reporting this, I'll see what I can do!

guidodev, 2020-09-11:
@zssherman You bet, I was scratching my head for similar reasons, because I didn't see anything in the 1.11.1 -> 1.11.2 release notes that would suggest this issue would crop up. FWIW, the testing I did was with pip3 and totally clean envs each time -- I even went to the trouble of creating a completely new user for each test. ;-) Maybe somehow it really is just related to dependencies -- that's actually why I included the P.S. (thinking perhaps it might help shed some light)

bmgxyz, 2020-09-14:
I ran into this issue as well with product 176. Unfortunately, rolling back isn't an option for me because I need product 176 for my problem, and support for product 176 arrived with v1.11.2.

I used `git bisect` to figure out where the bug arose. It seems to come from [this commit (c22ee73a)](https://github.com/ARM-DOE/pyart/commit/c22ee73a1af67e0b7113e2580046f0e2cebad276), which makes sense since the error complains about `NoneType`, and the whole point of that commit seems to be setting `elevation` to `None` in some cases.

I just started using `pyart` today, so it's not clear to me how to solve this problem correctly. However, when I apply this simple diff to `master`:

```diff
diff --git a/pyart/io/nexrad_level3.py b/pyart/io/nexrad_level3.py
index c4501155..2abc7a32 100644
--- a/pyart/io/nexrad_level3.py
+++ b/pyart/io/nexrad_level3.py
@@ -249,7 +249,7 @@ class NEXRADLevel3File(object):
         if self.msg_header['code'] in ELEVATION_ANGLE:
             elevation = struct.unpack('>h', hw30)[0] * 0.1
         else:
-            elevation = None
+            elevation = 0
         return elevation

     def get_volume_start_datetime(self):
```

I get the output I'd expect:

```
## You are using the Python ARM Radar Toolkit (Py-ART), an open source
## library for working with weather radar data. Py-ART is partly
## supported by the U.S. Department of Energy as part of the Atmospheric
## Radiation Measurement (ARM) Climate Research Facility, an Office of
## Science user facility.                              
##                                                         
## If you use this software to prepare a publication, please cite:
##                           
##     JJ Helmus and SM Collis, JORS 2016, doi: 10.5334/jors.119
                         
[[-85.92199  -85.92199  -85.92199  ... -85.92199  -85.92199  -85.92199 ]
 [-85.92199  -85.92194  -85.921906 ... -85.87929  -85.87924  -85.879196]
 [-85.92199  -85.921906 -85.92181  ... -85.8366   -85.8365   -85.83641 ]                                                                                                      
 ...
 [-85.92199  -85.92213  -85.922264 ... -86.05004  -86.050186 -86.05033 ]
 [-85.92199  -85.92208  -85.92218  ... -86.007385 -86.00748  -86.007576]
 [-85.92199  -85.92204  -85.92208  ... -85.96469  -85.964745 -85.96479 ]]
[[30.565    30.567245 30.569496 ... 32.626194 32.62844  32.630688]
 [30.565    30.567245 30.569496 ... 32.625874 32.62812  32.630367]
 [30.565    30.567245 30.569494 ... 32.624912 32.627155 32.629402]
 ...
 [30.565    30.567245 30.569487 ... 32.623306 32.62555  32.627796]
 [30.565    30.567245 30.569494 ... 32.624912 32.627155 32.629402]
 [30.565    30.567245 30.569496 ... 32.625874 32.62812  32.630367]]
```

which for all I know could be wrong, but seems reasonable to me. (The input was a recent product 176 file from KEVX, and those coordinates seem reasonable.)

Looking at it now, the commit that introduced support for product 176 came before this one, so maybe I can roll back to that point and use that until this is resolved.

I'm not sure what to do with the above information, but maybe someone with more experience can take this and run with it. I'd also be willing to try to solve it myself, if only someone would orient me.

Perhaps @dcedgren, the author of the above commit, could cast some light?

zssherman, 2020-09-14:
Thanks @bmgxyz, great catch also! After looking back at a previous discussion with dcedgren, we discussed whether to use None or 0.0, if I could find a compatibility issue, now looking at this, I think your fix is the way to go. I missed the compatibility issue with the coord retrievals. I will submit a PR for this, shortly.

zssherman, 2020-09-14:
The PR passed and has been accepted, let me know if it works. I'll have to get a micro release out for conda-forge soon

bmgxyz, 2020-09-14:
@zssherman Seems to work just fine on my machine. Thanks for the quick turnaround. :+1:

Of course, it's up to @guidodev whether this really solves their original problem, but FWIW I'm happy.

jleinonen, 2020-10-29:
For anyone with this problem before the fix makes it to their version, a quick workaround seems to be to just manually set `radar.elevation['data'] = 0.0`. After that `gate_longitude` and `gate_latitude` seem to work just fine.

zssherman, 2020-11-11:
Thanks again all for the suggested fixes, workarounds, etc. I did a new release, so closing this. Let me know if you run into any issues in the new release.

closed 2020-11-11 by zssherman