ARM-DOE/pyart #961: Inconsistent velocity dealiasing

lauratomkins, 2020-09-21:
I'm having an issue when dealiasing some radial velocity fields from NEXRAD data. I have 3 sequential velocity fields that have similar raw velocity fields and after dealiasing, 2 of the dealiased fields look fine but one of them seems to "over correct" the field. I've tried filtering the velocity before dealiasing (filtering values < 5 dBZ) but that makes the issue worse. I also tried despeckling before dealiasing but that didn't help either.

Here are the fields without filtering and without depeckling:
[image]

Here are the fields after filtering < 5 dBZ:
[image]

And here are the fields after despeckling:
[image]

Any ideas what's going on here or settings I might need to adjust?

lauratomkins, 2020-09-21:
Example code
```python
# read in file
radar = pyart.io.read_nexrad_archive('KBGM20200213_160420_V06')
#radar = pyart.io.read_nexrad_archive('KBGM20200213_161126_V06')
#radar = pyart.io.read_nexrad_archive('KBGM20200213_161832_V06')

# extract first sweep
radar = radar.extract_sweeps([1]) # extract velocity sweep

# filter velocity less than 5
#gatefilter = pyart.correct.GateFilter(radar)
#gatefilter.exclude_below('reflectivity', 5)

# despeckle field
gatefilter = pyart.correct.despeckle_field(radar, 'velocity')

# dealias velocity
corr_vel = pyart.correct.dealias_region_based(radar,vel_field='velocity',skip_along_ray=100,skip_between_rays=100,gatefilter=gatefilter,keep_original=False)
radar.add_field('dealiased_velocity', corr_vel, True)
```

lauratomkins, 2020-10-04:
Link with level2 files https://drive.google.com/file/d/1DDbImmX9anGtMTskBlxrhQE9aNNF3Nre/view?usp=sharing

rcjackson, 2020-10-05:
You usually need a *very* aggressive filter for clutter to have the velocity dealiasing to work correctly. It is vital that most, if not all, of the noise in the center is removed. I would suggest using a velocity texture filter. Higher values of velocity texture correspond to noise and multi-trip echoes, so this will help get rid of a lot of your noise in the center. 

You can get the velocity texture by using pyart.retrieve.calculate_velocity_texture. When you plot out the histogram you should see two peaks. There will be a peak that corresponds to hydrometeors and one for noise. The noisy peak will have a higher mode and will be Gaussian, while the peak corresponding to real echoes will be non-Gaussian and have a lower mode. I'd develop a threshold somewhere between the two peaks and try that as a filter. Once you've done the filtering, I'd then even do a despeckle that aggressively removes the center clutter.

lauratomkins, 2020-10-12:
Thanks for the advice @rcjackson ! 

I tried filtering by the velocity texture and despeckling but the algorithm still struggled (in different ways). I'm currently experimenting with using the `remove_small_objects` function from the scikit-image morphology package which seems to work really well.

closed 2020-10-12 by lauratomkins