ARM-DOE/pyart #962: Error reading NOAA P3 tail doppler radar data

mwilson14, 2020-09-28:
Hi everyone,

I've been working with some of the P3 radar data from TORUS using PyART, and when I first tried reading in one of the files I got this error:

[image]

I was able to fix it by changing the call to "np.rec.frombuffer" in line 47 of _sigmet_noaa_hh.py to "np.frombuffer", so I thought I'd put that here in case anyone else ran into the same issue.

zssherman, 2020-09-28:
Hi @mwilson14 yeah that looks like the submodule was changed so, I think your fix will work. I'll do a PR and author it as you so you get PR credit.

mwilson14, 2020-09-28:
Thanks! I'm just really glad that the fix turned out to be as small of a change as it was.

zssherman, 2020-11-11:
Was fixed in #963 closing.

closed 2020-11-11 by zssherman