ARM-DOE/pyart #967: TDWR Level II Files

HumphreysCarter, 2020-12-09:
Hello,

So I tired plotting Level II data from a TDWR but when I load the data in I get an AssertionError with ` pyart.io.read_nexrad_archive`. 

L2 file: https://noaa-nexrad-level2.s3.amazonaws.com/2020/12/09/TJFK/TJFK20201209_000314_V08

```
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<timed exec> in <module>

~/.local/lib/python3.7/site-packages/pyart/io/nexrad_archive.py in read_nexrad_archive(filename, field_names, additional_metadata, file_field_names, exclude_fields, include_fields, delay_field_loading, station, scans, linear_interp, **kwargs)
    186     available_moments = set([m for scan in scan_info for m in scan['moments']])
    187     interpolate = _find_scans_to_interp(
--> 188         scan_info, first_gate, gate_spacing, filemetadata)
    189 
    190     fields = {}

~/.local/lib/python3.7/site-packages/pyart/io/nexrad_archive.py in _find_scans_to_interp(scan_info, first_gate, gate_spacing, filemetadata)
    277                 # for proper interpolation the gate spacing of the scan to be
    278                 # interpolated should be 1/4th the spacing of the radar
--> 279                 assert spacing == gate_spacing * 4
    280                 # and the first gate for the scan should be one and half times
    281                 # the radar spacing past the radar first gate

AssertionError: 
```

zssherman, 2020-12-11:
Hi @HumphreysCarter, hmm, I'll have to think of a way around this. Seems the spacing for the file changes for some moments, but its not 1/4 it looks to be 1/2 which is failing the interpolation code. This is a relatively new files, so wonder if something change within the files themselves. Hmm... I'll have to think on this one.

zssherman, 2020-12-11:
Do you have another file possibly near that date range? I wonder if its file unique.

HumphreysCarter, 2020-12-11:
Yeah I tired several TDWR files, each had issues. Even with different sites, still had no luck.

https://noaa-nexrad-level2.s3.amazonaws.com/2020/12/09/TJFK/TJFK20201118_000425_V08 
https://noaa-nexrad-level2.s3.amazonaws.com/2020/12/09/TJFK/TJFK20201118_001625_V08  
https://noaa-nexrad-level2.s3.amazonaws.com/2020/12/09/TJFK/TJFK20201118_002825_V08  
https://noaa-nexrad-level2.s3.amazonaws.com/2020/12/09/TJFK/TJFK20201118_004024_V08  
https://noaa-nexrad-level2.s3.amazonaws.com/2020/12/09/TJFK/TJFK20201118_005224_V08
https://noaa-nexrad-level2.s3.amazonaws.com/2020/12/09/TJFK/TJFK20201118_001025_V08 
https://noaa-nexrad-level2.s3.amazonaws.com/2020/12/09/TJFK/TJFK20201118_002225_V08  
https://noaa-nexrad-level2.s3.amazonaws.com/2020/12/09/TJFK/TJFK20201118_003425_V08  
https://noaa-nexrad-level2.s3.amazonaws.com/2020/12/09/TJFK/TJFK20201118_004624_V08  
https://noaa-nexrad-level2.s3.amazonaws.com/2020/12/09/TJFK/TJFK20201118_005824_V08

zssherman, 2020-12-16:
@HumphreysCarter So it seems nexrad reader currently doesn't have the capability to read the TWDR datasets. And after discussion with @scollis, @dopplershift, does metpy have the capability to read TWDR? We were thinking of possibly doing a bridge to metpy for an io.

dopplershift, 2020-12-16:
@zssherman Yes, we handle the TDWR level 2 files. I'm happy to have you all bridge to us--but honestly, if you're already handling the Build 19 NEXRAD Level 2, there's not much to tweak to make TDWR work. We don't have any special code paths or anything, I think just a few asserts/checks that needed loosening.

I *think* [this commit](https://github.com/Unidata/MetPy/commit/f90f622aba148e13b8b587a19af006265786195b) is all we needed.

zssherman, 2020-12-16:
Makes sense. Yeah if that's all that requires reworking, I agree. I'll modify the pyart nexrad code with a implement of the linked commit to get this working. As always I appreciate it @dopplershift !

HumphreysCarter, 2020-12-17:
Thanks for the fix @dopplershift and @zssherman!