ARM-DOE/pyart #968: Gridding Hangs Up for Some Cf-Radial Files

dm0096, 2020-12-16:
Good afternoon,

I'm trying to grid two radar files from UAH ARMOR. Both are Cf-Radial .nc files. One file grids in ~9.5 seconds, and the other is still "gridding" as I write this (going on 25 minutes after execution). I've attached a test script and the two files in this link:

https://www.dropbox.com/sh/cc279ejqqcb3bs1/AAAADEbt6rStPEL7z5NN7yI3a?dl=0

To reproduce the problem: 
Run the attached script in a Py-ART environment. The first file should grid quickly, and the other should grid infinitely.

I can't discern what's different about these files besides that they're from two different times. The file starting at 2338 UTC is hanging up the gridding function. It should be netcdf4 compliant, or else Py-ART wouldn't read it at all. I created the file by converting from DORADE sweep files using RadxConvert. The only hypothesis I have is that Radx converting too many times has corrupted the file's data structure and slowed down the gridding function.

Thank you for your help,
Dean Meyer

scollis, 2020-12-16:
Interesting.

I would suggest tying a gatefilter 

https://arm-doe.github.io/pyart-docs-travis/API/generated/pyart.filters.GateFilter.html#pyart.filters.GateFilter

Throw in an exclude invalid https://arm-doe.github.io/pyart-docs-travis/API/generated/pyart.filters.GateFilter.exclude_invalid.html#pyart.filters.GateFilter.exclude_invalid

 

Pass that gatefilter (If I recall correctly it is gatefilters=(gf1, gf2, gfn) if you are gridding n radars so one radar is gatefilters=(gf1))

 

Why? Py-ART still adds masked/nans etc when its searching.. Now if you have a radar with no masking (eg down to the noisefloor) craft a gate filter for the signals you want to grid (we like using the texture of radial velocity as an indicator on when we are below the MDS)..

 

https://www.dropbox.com/sh/cc279ejqqcb3bs1/AAAADEbt6rStPEL7z5NN7yI3a?dl=0

dm0096, 2020-12-16:
I added gatefilters with identical constraints to each grid. The gatefilter, using exclude_invalid, does speed up the first grid by 2 seconds, but the second grid still won't finish. As a test, I then included an exclude_below for reflectivity < 20 dBZ, and that sped up the first grid by 3 seconds, but the second grid again didn't finish after several minutes.

scollis, 2020-12-16:
Interesting. Zach can you try gridding the slower file? 

-sent from a mobile device-

zssherman, 2020-12-16:
Hmm, yeah your map_gates_to_grid so no problems there, and your limits and shape aren't too high. Wonder what ppis prior to gridding look like. If anything looks off with the second file. I was going to say gatefilter as will, but yeah @scollis I'll try the slower file.

dm0096, 2020-12-16:
Here's PPIs of the files before gridding with only the exclude_invalid gatefilter applied. The right plot corresponds to the slow grid.
[image]

zssherman, 2020-12-16:
So I think I found a solution @dm0096 , so typically the radar latitude lon alt etc are 1 value and not arrays. I believe they are messing the init gate functions. Which throws me off more though is why the first radar still works with the same situation. What I did was:
radar2.altitude['data'] = np.array([radar2.altitude['data'][0]])
radar2.longitude['data'] = np.array([radar2.longitude['data'][0]])
radar2.latitude['data'] = np.array([radar2.latitude['data'][0]])
then reinitialized the gates:
radar2.init_gate_altitude()
radar2.init_gate_longitude_latitude()
radar2.init_gate_x_y_z()
and then grid and the second radar gridded in 6 seconds
bt = time.time()
grid2 = pyart.map.grid_from_radars(
            radar2, grid_shape=grid_shape,
            grid_limits=(zlim, ylim, xlim),
            grid_origin=origin, 
            fields=fields,
            gridding_algo='map_gates_to_grid', 
            grid_origin_alt=0.0,
            weighting_function='Barnes2')
print(f'{time.time()-bt} seconds to grid radar')

Again, i'm confused on why radar 1 still gridded, so my guess is the underlying inits got thrown off in radar 2, but @dm0096 let me know if my solution works on your end.

dm0096, 2020-12-17:
@zssherman It works! Going forward, I'll add failsafe measures in my code to use your procedure whenever the lat/lon/alt data comes in arrays. 

Thanks everyone!!

dm0096, 2020-12-17:
I should expand on my last comment. After further experimentation, it's not necessarily that the gridding hangs up whenever the lat/lon/alt data are in arrays, it's that it hangs _when they're MaskedArrays._ Retrieving only the un-masked values solves the issue as well. I also masked some disturbing altitudes (-9998 m) first.

zssherman, 2020-12-17:
@dm0096 Ahh yep that now makes sense! Thanks for letting me know, I was scratching my head on that one.

zssherman, 2020-12-17:
I'm going to close this for now @dm0096, if something arises I can always reopen.

closed 2020-12-17 by zssherman