ARM-DOE/pyart #970: How to cutoff radar moments outside a specific range value?

jaypotnis, 2020-12-30:
Hello all,
I am new to radar data processing and have recently started using pyart. I have written a code for plotting radar moments on a cartopy map. I am using the pyart.graph.RadarMapDisplay.plot_ppi_map function to plot radar moments on the map. Below is a snippet of the code I am using for plotting.

`display.plot_ppi_map('reflectivity', sweep= sweep, ax=axeses[0][0], vmin=0, vmax=70,
                     min_lon=long-1.2, max_lon=long+1.2, min_lat=lat-1.2, max_lat=lat+1.2, resolution='10m', projection=projection,
                     fig=fig, lat_0=radar.latitude['data'][0], title = 'Reflectivity', colorbar_label='Z (dBZ)',
                     lon_0=radar.longitude['data'][0], mask_outside=True, alpha=1)

display.plot_range_ring(100., line_style='k--')
display.plot_point(radar.longitude['data'][0], radar.latitude['data'][0],'ko')`

These are the type of plots I am getting after plotting.

[image]

As you can see in the plot, there is reflectivity appearing outside the 100 km range ring. I wanted to cut all the reflectivity values outside the 100 km range ring. Is there a way to do that? This may seem a very basic issue, but I am unable to find a solution for it.

Thanks for helping out.

zssherman, 2020-12-30:
Hi @jaypotnis, you can try to index and mask a copy of the radar reflectivity with your range attribute,
something like
rng = radar.range['data']
radar.add_field_like('reflectivity', 'reflectivity_copy',
                                radar.fields['reflectivity']['data'].copy())
radar.fields['reflectivity_copy']['data'][:, range > 100000]

somewhere along those lines

jaypotnis, 2021-01-02:
That worked. Thank you very much for your help.

closed 2021-01-02 by jaypotnis