ARM-DOE/pyart #972: Problems with Cartopy and pyart

wxwolff, 2021-01-05:
I am suddenly having trouble getting RadarMapDisplay to draw radar display on Cartopy map on my Mac. The code below shows my computer, OS and the versions of arm_pyart, Cartopy and Matplotlib.  This is from a fresh install of anaconda and subsequent install of arm_pyart and Cartopy using conda.

import numpy as np
import matplotlib.pyplot as plt
import pyart
import cartopy
import cartopy.crs as ccrs
import cartopy.feature as cfeature
from cartopy.mpl.gridliner import LONGITUDE_FORMATTER, LATITUDE_FORMATTER

'''
MacPro 2020 / Mac OS 10.15.7
​
conda list | grep 'pyart\|matplotlib\|cartopy'
arm_pyart                 1.11.3           py38hf3d72b9_0    conda-forge
cartopy                   0.17.0          py38h9bcff04_1015    conda-forge
matplotlib                3.3.2                         0  
matplotlib-base           3.3.2            py38h181983e_0  
'''
​
​
import numpy as np
import matplotlib.pyplot as plt
import pyart
import cartopy
import cartopy.crs as ccrs
import cartopy.feature as cfeature
from cartopy.mpl.gridliner import LONGITUDE_FORMATTER, LATITUDE_FORMATTER
#import urllib
'''
arm_pyart                 1.11.3           py38hf3d72b9_0    conda-forge
cartopy                   0.17.0          py38h9bcff04_1015    conda-forge
matplotlib                3.3.2                         0  
matplotlib-base           3.3.2            py38h181983e_0  
'''

if __name__ == "__main__":

    #file = '/Volumes/RAID/Data/NPOL/16bit/PPI/product_raw/np1200430200017.RAWAJC2.gz'
    file = '/Volumes/RAID/Data/NPOL/iris/PPI/np1200430165902.RAWAJNN.gz'

    #url = 'https://rain.gsfc.nasa.gov/distro/BLOSSOM/np1200430200017.RAWAJC2.gz'
    #file = 'npol_test.gz'
    #print('Downloading ' + file + '...')
    #urllib.request.urlretrieve(url, file)
    
    print('Loading radar...')
    radar = pyart.io.read(file, file_field_names=True)
    radar_lat = radar.latitude['data'][0]
    radar_lon = radar.longitude['data'][0]
    print(radar_lon, radar_lat)
    print(radar.fields.keys())
    
    projection = ccrs.LambertConformal(radar_lon, radar_lat)
    display = pyart.graph.RadarMapDisplay(radar)
    
    fig = plt.figure(figsize=(9,6))
    mx = fig.add_subplot(111, projection=projection)
    display.plot('DBZ', sweep=0, vmin=0, vmax=65, cmap='pyart_NWSRef',
                 transform=projection, ax=mx)
    
    display.plot_range_rings(range(0,151,50))
    map_limit = [ -77, -73, 36.5, 40] # NPOL

    print('Done.')

Running from jupyter-lab...

## You are using the Python ARM Radar Toolkit (Py-ART), an open source
## library for working with weather radar data. Py-ART is partly
## supported by the U.S. Department of Energy as part of the Atmospheric
## Radiation Measurement (ARM) Climate Research Facility, an Office of
## Science user facility.
##
## If you use this software to prepare a publication, please cite:
##
##     JJ Helmus and SM Collis, JORS 2016, doi: 10.5334/jors.119

Loading radar...
/Users/wolff/anaconda3/lib/python3.8/site-packages/pyart/io/sigmet.py:131: RuntimeWarning: invalid value encountered in sqrt
  sigmet_data, sigmet_metadata = sigmetfile.read_data(full_xhdr=full_xhdr)
-75.34166989848018 38.262630086392164
dict_keys(['DBZ', 'VEL', 'WIDTH', 'ZDR', 'KDP', 'PHIDP', 'SQI', 'RHOHV'])
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-1-8fba4740e339> in <module>
     30     fig = plt.figure(figsize=(9,6))
     31     mx = fig.add_subplot(111, projection=projection)
---> 32     display.plot_ppi_map('DBZ', sweep=0, vmin=0, vmax=65, cmap='pyart_NWSRef',
     33                          projection=projection, ax=mx)
     34 

~/anaconda3/lib/python3.8/site-packages/pyart/graph/radarmapdisplay.py in plot_ppi_map(self, field, sweep, mask_tuple, vmin, vmax, cmap, norm, mask_outside, title, title_flag, colorbar_flag, colorbar_label, ax, fig, lat_lines, lon_lines, projection, min_lon, max_lon, min_lat, max_lat, width, height, lon_0, lat_0, resolution, shapefile, shapefile_kwargs, edges, gatefilter, filter_transitions, embelish, raster, ticks, ticklabs, alpha)
    269         if norm is not None:  # if norm is set do not override with vmin/vmax
    270             vmin = vmax = None
--> 271         pm = ax.pcolormesh(x * 1000., y * 1000., data, alpha=alpha,
    272                            vmin=vmin, vmax=vmax, cmap=cmap,
    273                            norm=norm, transform=self.grid_projection)

~/anaconda3/lib/python3.8/site-packages/cartopy/mpl/geoaxes.py in pcolormesh(self, *args, **kwargs)
   1457                              ' consider using PlateCarree/RotatedPole.')
   1458         kwargs.setdefault('transform', t)
-> 1459         result = self._pcolormesh_patched(*args, **kwargs)
   1460         self.autoscale_view()
   1461         return result

~/anaconda3/lib/python3.8/site-packages/cartopy/mpl/geoaxes.py in _pcolormesh_patched(self, *args, **kwargs)
   1489         allmatch = (shading == 'gouraud')
   1490 
-> 1491         X, Y, C = self._pcolorargs('pcolormesh', *args, allmatch=allmatch)
   1492         Ny, Nx = X.shape
   1493 

TypeError: _pcolorargs() got an unexpected keyword argument 'allmatch'
/Users/wolff/anaconda3/lib/python3.8/site-packages/cartopy/mpl/geoaxes.py:387: MatplotlibDeprecationWarning: 
The 'inframe' parameter of draw() was deprecated in Matplotlib 3.3 and will be removed two minor releases later. Use Axes.redraw_in_frame() instead. If any parameter follows 'inframe', they should be passed as keyword, not positionally.
  return matplotlib.axes.Axes.draw(self, renderer=renderer,

zssherman, 2021-01-05:
Hi @wxwolff I believe this was a bug in Cartopy 0.17, can you update to 0.18 and let me know if that works?

zssherman, 2021-10-11:
@wxwolff Still having issues with this? If not, can close this.

closed 2021-12-16 by zssherman