ARM-DOE/pyart #974: Weird ZDR Data Issues When Reading NOMADS Level II Data

mwilson14, 2021-01-06:
Hi pyart folks!

So I've been using pyart to read real-time level II data from the nomads level II feed (https://nomads.ncep.noaa.gov/pub/data/nccf/radar/nexrad_level2/), and I've noticed that pyart recently started having problems reading in the ZDR data from these level II files. Where before the data values were within the ZDR valid data range of ~-8 to 8 dB, now the maximum values I'm seeing for a given scan are around -5 dB. Additionally, the dimensions of the ZDR data seem strange, since there appears to be blank data every other range gate. I've been able to read the same radar files in GR2 with no issues, and pyart could read data from the nomads feed fine earlier in 2020, so I'm not entirely sure what's going on. I've attached a .zip file with an example of the weird level II data and a jupyter notebook demonstrating the issue--let me know if you've got any questions or insights into what's happening!

[Weird ZDR Issue.zip](https://github.com/ARM-DOE/pyart/files/5777468/Weird.ZDR.Issue.zip)

zssherman, 2021-01-07:
Hi @mwilson14 , interesting I will look into this. Have you updated pyart recently. We did some changes to the nexrad reader about a week ago. So if you did that might help narrow it down that maybe our changes could have broke something.

mwilson14, 2021-01-07:
I think I last updated it earlier this fall, so I'll update it today and see if that fixes the issue.

mwilson14, 2021-01-07:
Updating to the latest pyart release appears to have fixed the problem. Thanks!

zssherman, 2021-01-07:
Glad to hear @mwilson14 !