ARM-DOE/pyart #975: Issue creating labels RadarMapDisplay

William0251, 2021-01-08:
Hello,

I am new to python and I am try trying to create some symbols in my PPI plot so that I can represent the locations of the radars that I am using in my research. This seemed to me as an easy solve, but I can't figure out how to get further. For my code, I used the example code for creating a PPI plot on a cartopy map(https://arm-doe.github.io/pyart/source/auto_examples/plotting/plot_ppi_with_rings.html#sphx-glr-source-auto-examples-plotting-plot-ppi-with-rings-py)
The data that I am using is in netCDF4 format and there seems to be nothing wrong with it, since the PPI can be plotted without the locations of the radars in it. A picture of the PPI that I got now can be seen below:
[image]

The label of one radar is already plotted using the plot_point command.

I tried to use 'RadarMapDisplay.plot_label' to plot the locations of the other radars, but for some reason it doesn't read the coördinates properly. The code that I am using:

%% Author: Jason Hemedinger

import numpy as np
import matplotlib.pyplot as plt
import cartopy.crs as ccrs
import pyart

%% Read in the file, create a RadarMapDisplay object
filename = "E:\\Rijnmond Data\\level2_archive_RIJNMOND_2020-09-26T23-54-20Z.nc"
radar = pyart.io.read(filename)
display = pyart.graph.RadarMapDisplay(radar)

%% Setting projection and ploting the second tilt
projection = ccrs.LambertConformal(central_latitude=radar.latitude['data'][0],
                                   central_longitude=radar.longitude['data'][0])

fig = plt.figure(figsize=(10,10))
display.plot_ppi_map('equivalent_reflectivity_factor_HH', 1, vmin=-20, vmax=45,
                     min_lon=3.8, max_lon=5.1, min_lat=51.55, max_lat=52.3,
                     lon_lines=np.arange(3.7, 5., .3), resolution='10m',
                     lat_lines=np.arange(51.5, 52.3, .3), projection=projection,
                     fig=fig, lat_0=radar.latitude['data'][0],
                     lon_0=radar.longitude['data'][0], cmap='jet')

%% Plot range rings at 10, 20, 30, 40km
display.plot_range_ring(10., line_style='k-')
display.plot_range_ring(20., line_style='k-')
display.plot_range_ring(30., line_style='k-')
display.plot_range_ring(40., line_style='k-')

%% Plot cross hairs
display.plot_line_xy(np.array([-40000.0, 40000.0]), np.array([0.0, 0.0]), line_style='k-')
display.plot_line_xy(np.array([0.0, 0.0]), np.array([-40000.0, 40000.0]),line_style='k-')

%% Indicate the radar location with a point
display.plot_point(radar.longitude['data'][0], radar.latitude['data'][0],label_text='Rijnmond')

%% label locations of the other radars (here it goes wrong)
display.plot_label('MRR',(51.99617,4.37868), symbol='r+')
display.plot_label('Herwijnen',(51.8368788, 5.138272),symbol='r+')

plt.rc('font', size = 18)
plt.xlabel('Longitude')
plt.ylabel('Latitude')

plt.show()

I tried to use the radar.longitude and radar.latitude commands before my coordinates in display.label, but when I did this I got an error. Does somebody might know what I am doing wrong? 

The labels that I want to obtain are similar to the ones in this plot:
[image]

William0251, 2021-01-11:
If there is any information missing or if something is unclear, please let me know!

zssherman, 2021-01-11:
Hi @William0251, would plot_point still work with label text for the other two radars, similar to the first radar point you plotted or do you only want the label to show? I don't see the plot_label function in radarmapdisplaycartopy, my guess is projections are messing with it as it wasn't adapted from what I see. I'll have to see if I can get the plot_label to work with cartopy as well, or see if its a bug.

William0251, 2021-01-12:
Thanks for the reply. I substituded plot_label for plot_point and it worked! I runned the code again with plot_label, to check wether it worked because I restarted the code, but plot_label still doesn't work. I simply wanted to highlight the locations of the radars, so plot_point is actually perfect. Thanks for the help!

The figure:
[image]

Part of the code that made it work:

lats = [51.99617, 51.924234, 51.8368788]
lons = [4.37868, 4.471955, 5.138272]

# Indicate the radar location with a point
display.plot_point(radar.longitude['data'][0], radar.latitude['data'][0], label_text='Rijnmond')
display.plot_point(lons[0], lats[0], label_text='MRR')
display.plot_point(lons[2], lats[2], label_text='Herwijnen')

# display.plot_label('MRR',(lats[0], lons[0]), symbol='r+')
# display.plot_label('Herwijnen',(lats[2], lons[2]), symbol='r+')

closed 2021-01-12 by William0251