ARM-DOE/pyart #976: assigning elevation angles and surveillance versus doppler designation to each sweep in NEXRAD L2 data

tjturnage, 2021-01-17:
Hello,
With the commonly used SAILS (Supplemental Adaptive Intra-volume Low-level Scan) and MRLE ((Mid-Volume Rescan of Low-level Elevations) settings for Volume Coverage Patterns in NEXRAD, it's often difficult to know which sweeps pertain to which elevation angles. Furthermore,  it is challenging to know whether sweeps at the lower elevation angles are associated with a Contiguous Surveillance (CS) scan or with a Contiguous Doppler (CD. This is important for extracting the appropriate datasets and for plotting the maximum available 0.5 degree cuts for animations, etc.

I therefore suggest that a method be added that creates a dictionary tying together "sweep_number" and "elevation" arrays along with information about whether each sweep is CS, CD, or some other type of cut (such as Batch).  I have some rough code for this and can start a PR if there is interest.

Thanks!
TJ

zssherman, 2021-01-19:
I think this would be of interest @tjturnage! If you had some code, I would say a PR would be great and I can try to help where need be.

mgrover1, 2024-05-10:
@tjturnage - following up here, is this something you are still interested in?

tjturnage, 2024-05-10:
I am! And I can submit a PR unless you or someone else is about to tackle it.

mgrover1, 2024-05-10:
A PR here would be great 😄

dopplershift, 2024-05-10:
Adding this might help with #1559.