ARM-DOE/pyart #977: `pyart.aux_io.read_gamic()` failing with h5py 3.1.0

cclopes, 2021-02-18:
When I try to open a file with `pyart.aux_io.read_gamic()` in an environment with h5py 3.1.0, the following error occurs:

```python
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-22674a19b649> in <module>
----> 1 pyart.aux_io.read_gamic("../dados/radar/SR/SRO-250--2017-03-14--18-30-22.mvol")

~/miniconda3/envs/tutoriais-python/lib/python3.8/site-packages/pyart/aux_io/gamic_hdf5.py in read_gamic(filename, field_names, additional_metadata, file_field_names, exclude_fields, include_fields, valid_range_from_file, units_from_file, pulse_width, **kwargs)
    135     # sweep_type
    136     scan_type = gfile.raw_scan0_group_attr('what', 'scan_type').lower()
--> 137     scan_type = scan_type.decode('utf-8')
    138     # check that all scans in the volume are the same type
    139     if not gfile.is_file_single_scan_type():

AttributeError: 'str' object has no attribute 'decode'
```

<details>
<summary>Environment that caused the error</summary>

```bash
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
arm_pyart                 1.11.3           py38h5c078b8_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.1                      py_0    conda-forge
boost-cpp                 1.74.0               hc6e9bd1_2    conda-forge
brotlipy                  0.7.0           py38h497a2fe_1001    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.17.1               h36c2ea0_0    conda-forge
ca-certificates           2020.12.5            ha878542_0    conda-forge
cached-property           1.5.1                      py_0    conda-forge
cairo                     1.16.0            h7979940_1007    conda-forge
cartopy                   0.18.0          py38h2516ec2_10    conda-forge
certifi                   2020.12.5        py38h578d9bd_1    conda-forge
cffi                      1.14.5           py38ha65f79e_0    conda-forge
cfitsio                   3.470                hb418390_7    conda-forge
cftime                    1.4.1            py38h5c078b8_0    conda-forge
chardet                   4.0.0            py38h578d9bd_1    conda-forge
cryptography              3.4.4            py38h3e25421_0    conda-forge
curl                      7.71.1               he644dc0_8    conda-forge
cycler                    0.10.0                     py_2    conda-forge
decorator                 4.4.2                      py_0    conda-forge
deprecation               2.1.0              pyh9f0ad1d_0    conda-forge
expat                     2.2.10               h9c3ff4c_0    conda-forge
fontconfig                2.13.1            hba837de_1004    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
freexl                    1.0.5             h516909a_1002    conda-forge
gdal                      3.2.1            py38hc0b2d6b_5    conda-forge
geos                      3.9.0                h9c3ff4c_0    conda-forge
geotiff                   1.6.0                h2b14fbe_4    conda-forge
gettext                   0.19.8.1          h0b5b191_1005    conda-forge
giflib                    5.2.1                h36c2ea0_2    conda-forge
h5netcdf                  0.10.0             pyhd8ed1ab_0    conda-forge
h5py                      3.1.0           nompi_py38hafa665b_100    conda-forge
hdf4                      4.2.13            h10796ff_1004    conda-forge
hdf5                      1.10.6          nompi_h6a2412b_1114    conda-forge
icu                       68.1                 h58526e2_0    conda-forge
idna                      2.10               pyh9f0ad1d_0    conda-forge
ipykernel                 5.4.3            py38h81c977d_0    conda-forge
ipython                   7.20.0           py38h81c977d_2    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
jedi                      0.18.0           py38h578d9bd_2    conda-forge
jpeg                      9d                   h36c2ea0_0    conda-forge
json-c                    0.15                 h98cffda_0    conda-forge
jupyter_client            6.1.11             pyhd8ed1ab_1    conda-forge
jupyter_core              4.7.1            py38h578d9bd_0    conda-forge
[... 253 lines cut ...]
xarray                    0.16.2             pyhd8ed1ab_0    conda-forge
xerces-c                  3.2.3                h9d8b166_2    conda-forge
xmltodict                 0.12.0                     py_0    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h516909a_0    conda-forge
xorg-libsm                1.2.3             h84519dc_1000    conda-forge
xorg-libx11               1.6.12               h516909a_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h516909a_0    conda-forge
xorg-libxrender           0.9.10            h516909a_1002    conda-forge
xorg-renderproto          0.11.1            h14c3975_1002    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
zeromq                    4.3.4                h9c3ff4c_0    conda-forge
zlib                      1.2.11            h516909a_1010    conda-forge
zstd                      1.4.8                ha95c52a_1    conda-forge
```

</details>

Tested with these GAMIC-HDF5 files: [.mvol](https://github.com/LabNuvens/tutoriais-python/blob/main/dados/radar/SR/SRO-250--2017-03-14--18-30-22.mvol), [.HDF5](https://github.com/LabNuvens/tutoriais-python/blob/main/dados/radar/SIPAM/RADL08061720140103111200.HDF5)

I believe this issue is related to h5py because [this function](https://github.com/LabNuvens/tutoriais-python/blob/main/1.Radar/read_brazil_radar_py3.py) that uses h5py to read files is also failing because of a string that is read as `"b'[some string]'"` in the older version and `"[some string]"` in the newer version.

zssherman, 2021-02-18:
@cclopes Thank you for the information! Yeah seems they are possibly decoding during reading and we are duplicating it in the newer versions. I'll add a try and except at the moment, for a workaround to work with earlier version. I'll submit a PR. Thanks again for bringing this to our attention

kmuehlbauer, 2021-02-18:
This breaking change in `h5py>=3.0` has been a major pain for several packages. I think this will not be the only place where this might surface.

zssherman, 2021-02-18:
@kmuehlbauer Ah gotcha, yeah we use hdf5 in alot of the aux io s , so have to do some digging to see where else this will arise. I'm assuming its not a bug on their part, just we have to now remove decoding on our end?

kmuehlbauer, 2021-02-18:
@zssherman No, it's not a bug in that sense. The new string handling is described here:

https://docs.h5py.org/en/stable/strings.html#reading-strings

> String data in HDF5 datasets is read as bytes by default: bytes objects for variable-length strings, or numpy bytes arrays ('S' dtypes) for fixed-length strings. Use Dataset.asstr() to retrieve str objects.
> 
> Variable-length strings in attributes are read as str objects. These are decoded as UTF-8 with surrogate escaping for unrecognised bytes.

So in general all strings are represented as bytes, but not attributes vlen_strings, those are `str`.  Hope you find your way through this!

zssherman, 2021-02-18:
@kmuehlbauer Makes sense and that link was helpful, I think I have a fix. Thanks again!

zssherman, 2021-02-18:
@cclopes I implemented a fix and tested it with the file you provided. I accepted the PR and should be available using git. Thanks again for bringing this to our attention.

cclopes, 2021-02-18:
That's great! Thanks for the quick fix 😄

closed 2021-02-18 by cclopes