ARM-DOE/pyart #981: radar location shifting in pyart module

avishekray, 2021-03-02:
I am using the below code for ppi ploting 

import pyart
import matplotlib.pyplot as plt
import netCDF4
import numpy as np
import cartopy.crs as ccrs

filename = 'KASPR_20190609-115944_vpp.nc'
radar=pyart.io.read(filename)

projection = ccrs.LambertConformal(central_latitude=19.0,
                                   central_longitude=80.0)
fig = plt.figure(figsize=(6,6))
display.plot_ppi_map('DBZ', 0,vmin=-30, vmax=40,
                     min_lon=73.5, max_lon=74.2, min_lat=17.9, max_lat=18.9,
                      resolution='10m',
                      projection=projection,
                     fig=fig, lat_0=19.0,
                     lon_0=80.0,embelish=True,
            ticks=None, ticklabs=None, raster=False)

where DBZ is a variable. As the radar location lat lon (18.5N,75E) is wrong in file so I have to change the radar location lat lon to (19.0N,80.0E). But it is not reflecting in plot. Radar location remains same in plot as in file. How to resolve this?

scollis, 2021-03-02:
That changes the location of the plot.. not the radar

You need to change radar location in the object itself

https://github.com/ARM-DOE/pyart/blob/master/pyart/core/radar.py#L46

 

note that these are dictionaries and the radar.longitude[‘data’] is an array to allow for moving platforms

replace the existing values with your location values

important do this right after loading…

avishekray, 2021-03-03:
I have changes the code in radar.py in below bold italic line

def _gate_lon_lat_data_factory(radar, coordinate):
    """ Return a function which returns the geographic locations of gates. """
    def _gate_lon_lat_data():
        """ The function which returns the geographic locations gates. """
        x = radar.gate_x['data']
        y = radar.gate_y['data']
        projparams = radar.projection.copy()
        if projparams.pop('_include_lon_0_lat_0', False):
            **_projparams['lon_0'] = 80
            projparams['lat_0'] = 19_**
        geographic_coords = cartesian_to_geographic(x, y, projparams)
        # set the other geographic coordinate
        if coordinate == 0:
            radar.gate_latitude['data'] = geographic_coords[1]
        else:
            radar.gate_longitude['data'] = geographic_coords[0]
        return geographic_coords[coordinate]
    return _gate_lon_lat_data

But radar..longitude[‘data’] is not showing 8. It is showing radar on which is in the file. So when I am plotting reflectivity, plot is coming as lat lon in the file. It is not updating my given lat lon. Can you please suggest how to resolve that?

zssherman, 2021-03-03:
Hi @avishekray I wouldn't change the function itself. I believe @scollis was just pointing you to the reset functions to use inside radar.py. What you want to do is as soon as you read in your radar:
radar = pyart.io.read('filepath')
radar.latitude['data'] = np.array([19.0])
radar.longitude['data'] = np.array(['80.0'])
radar.init_gate_longitude_latitude()
radar.init_gate_altitude()

That should recalculate the gate latitudes and longitudes for the radar based on the new lat and lon you provided.

avishekray, 2021-03-03:
Thanks, @scollis, and  @zssherman for your quick reply and for solving the issue.
I have understood where I am doing the mistake and thanks @zssherman for showing your code.

closed 2021-03-03 by avishekray