ARM-DOE/pyart #982: Python 3.9 support

vlouf, 2021-03-11:
Is there a Py-ART a release supporting Python 3.9 in the work? What can be done to help you make this possible?

zssherman, 2021-03-11:
@vlouf I believe all c files have been cythonize for pyart for python 3.9. If I remember correctly, I haven't been able to add python 3.9 support because some dependecies haven't updated yet, the last time the conda forge bot ran. So I can try to rerender the bot on arm pyart feedstock to see if it works now, and if so then it should be able to support 3.9, but yeah conflicts of dependecies where holding it back:
https://github.com/conda-forge/arm_pyart-feedstock/pull/39

zssherman, 2021-03-11:
I triggered the bot to rerun, I'll see what the CI does.

zssherman, 2021-03-11:
So there was a little bit of cython fixing I missed in a previous PR, but everything is good, feedstock and github CI has been updated. @vlouf Say give it like 20 mins or so and Python3.9 release for Py-ART should show up on conda-forge. I close this once its up and confirmation it works.

closed 2021-04-20 by vlouf