ARM-DOE/pyart #987: In the bugsuse / pyart which forked from pyart,I cant import pyart.io.sband_archive

niguanne-coder, 2021-04-10:
cannot import name '_fast_interpolate_scan_4' from 'pyart.io.nexrad_interpolate' (C:\Users\ghd\anaconda3\lib\site-packages\pyart\io\nexrad_interpolate.cp38-win_amd64.pyd)?

zssherman, 2021-04-13:
@niguanne-coder Hmm, we updated the nexrad code a couple of months ago.  The maintainer of that fork, I would ask if they updated there fork to the changes we have done recently? Or possibly a conflict? I don't have much say over that fork, but I can try to resolve this if the fork is updated.

zssherman, 2021-04-13:
If your locally installing it too, and the fork has been updated but new cython files haven't been generated, might need to do that as well.

closed 2021-10-11 by zssherman