ARM-DOE/pyart #993: Unidata removed utime class from cftime

amarouane-ABDELHAK, 2021-05-24:
Unidata removed utime class from cftime which will break this line https://github.com/ARM-DOE/pyart/blob/8c838bc823436e2397253fb66b06d47c42c43816/pyart/aux_io/sinarame_h5.py#L13
https://pypi.org/project/cftime/1.5.0/
`Some legacy classes and functions removed (utime, JulianDayFromDate and DateFromJulianDay). The functionality of JulianDayFromDate and DateFromJulianDay is now available from the methods cftime.datetime.toordinal and cftime.datetime.fromordinal.`

zssherman, 2021-05-24:
Thanks @amarouane-ABDELHAK for posting this and the info as well! I'll be working on a fix shortly and will have a PR soon

zssherman, 2021-05-24:
Looks like you as well posted in https://github.com/Unidata/cftime/issues/241, which address the question i had as well, the replacement for utime. If i can't find anything in the changelogs, hopefully there is a response and i will implement the replacement.

PR #994 (2021-05-25): FIX: Fix for cftime 1.5 release.
Fixes the removal of cftime utime legacy code, reproduces similar cftime use in other modules. Fixes #993
closed 2021-05-25 by zssherman