ARM-DOE/pyart #996: write_grid

vgalligani, 2021-05-26:
Hi! I'm trying to save grids I made through pyart.map.grid_from_radars. I've noticed that a few issues have been closed regarding writing grids and netcdf, but I have another issue. 

I've added a few prints inside the code and I have noticed that the netcdf variable that is saved inside write_grid through _create_ncvar() has dimensions of (720,1,720,781) for ('time', 'z', 'y', 'x'). This is incorrect, I'm expecting (1, 1, 720, 781). It's making 'time' to have 720 when it should be 1. 
Inside write_grid, field_dic['data'] has the right dimensions, but somehow inside _create_ncvar()  things get funny and I'm not following the code well. I can't narrow the bug down. 

Thank you !

zssherman, 2021-05-26:
Hmm, i'm not able to reproduce, do you have a test file I can try out to see if I can reproduce, what is the shape of your radar.time array?

vgalligani, 2021-05-26:
Hi! You are correct, sorry. My grid.time['data'] array is actually 720. I was certain I was working with dimension 1. 
I am using TINT and it seems they assume that for each sweep the time dimension is always 1. I'll see how to work around this. sorry and thanks!!!

zssherman, 2021-05-26:
No worries, are you hard setting the grid.time['data']? If not trying to see how its being set to 720.

zssherman, 2021-05-26:
Time gets set to:
  time = get_metadata('grid_time')
time['data'] = np.array([first_radar.time['data'][0]])
time['units'] = first_radar.time['units']

Which is where the single time is being generated

vgalligani, 2021-05-26:
Thanks!
Do you suggest chaning that to the full 720 times? Wouldn't that make a single PPI into various grids? meaning a grid for each set of gates that have the same time?

zssherman, 2021-05-26:
No, I would have the single time, just was seeing how the 720 was overwriting the code I posted above.

vgalligani, 2021-05-26:
The grid seems to work, it seems to get modified when I write it with pyart.io.write_grid(name, grid) 
inside_create_ncvar(), the first lines 

data = dic['data'] has the right shape 
data.shape =  (1, 1, 720, 781) 

but then, towards then end

ncvar = dataset.createVariable(name, dtype, dimensions, **kwargs)
ncvar is returned to the 720 time dimension

ncvar.shape =  (720, 1, 720, 781) 

this is what confuses me

zssherman, 2021-05-26:
The write also outputs radar_time as well which could be the number. After writing the grid to make sure though, with your written grid, do:
import netCDF4
data = netCDF4.Dataset('grid_name')
the print(data) and it should have your dimensions there

vgalligani, 2021-05-26:
Ok, so the grid written has the right dimensions, the problem was actually in the read_grid(name)

in the lines where

if field_dic['data'].shape == field_shape_with_time:
         field_dic['data'].shape = field_shape
         fields[field] = field_dic
else:
         bad_shape = field_dic['data'].shape
            warnings.warn(
                'Field %s skipped due to incorrect shape %s'
                % (field, bad_shape))

with the following for the Zh in my grid: 
field_shape_with_time = (1, 1, 720, 781)
field_dic[data].shape = (720, 1, 720, 781)

so I replaced
    field_shape = tuple([len(dset.dimensions[d]) for d in [,'z', 'y', 'x']])
    field_shape_with_time = (1, ) + field_shape   

with 
    field_shape = tuple([len(dset.dimensions[d]) for d in ['time','z', 'y', 'x']])
    field_shape_with_time =  field_shape  
    
I think this has fixed my issue.