ARM-DOE/pyart #1000: No module named 'pyart.__check_build._check_build

wangjin6912, 2021-07-03:
I'm trying to import pyart, however when I attempt to do so I receive the following:
ImportError: No module named 'pyart.__check_build._check_build',It seems that Py-ART has not been built correctly.
I'm fairly sure that pyart has been correctly built as I managed to run ”conda env create -f environment.yml “ without any incident.
Might anybody have an idea on how to fix this?

taort, 2021-08-03:
I met the same problem. Have you solved it ?

zssherman, 2021-08-03:
Hello, is this the environment.yml from the main repository here:https://github.com/ARM-DOE/pyart/blob/master/environment.yml ? Or is it a version from a forked version or modified? I haven't been able to reproduce yet. Can you post the content of the environment.yml file your using? And do a conda list as well

zssherman, 2021-08-04:
I was finally able to reproduce the issue @wangjin6912 @taort. I'll see if I can find a solution or a fix to whats going on.

zssherman, 2021-08-04:
@wangjin6912 @taort Are either of you importing the conda version of pyart from within a pyart github directory? The conda builds look fine, but when I reproduce it is when im importing conda pyart from within the pyart directory. The check build doesn't pickup conda install. Do you both have a local version? If so, try import pyart not inside the pyart directory. Also, if you are doing development on pyart for any reason, dont conda install it, pip install -e . in development mode.