ARM-DOE/pyart #1004: GAMIC's volume might not have puse_width value

ReallyNiceGuy, 2021-07-27:
The format of GAMIC's volume states that pulse_width is an optional information.
PYART will raise an error in this situation, but it would be better if it didn't. I would like to know which value should be used instead.
I could put an explicit "None" there.
No code should break with this change, as the error will propagate to the usage point instead of the loading point.

rcjackson, 2021-10-11:
What is the pulse width of your radar in degrees? That should be the value that goes there, which you may have to manually specify.

rcjackson, 2021-10-11:
Could you send us an example file without a pulse_width field? I suspect we just need to put a default value of none in the reader in gamic_hdf5.py if there is no field found. @zssherman and I will take a look.

mgrover1, 2022-07-06:
@zssherman can we close this via #1014 ?

zssherman, 2022-07-06:
Yep closing!

closed 2022-07-06 by zssherman