ARM-DOE/pyart #1011: Time error in creating basic ingest

syedhamidali, 2021-09-12:
Dear Pyart Team,
I was trying to create basic ingest, but it is encountering some errors in time. Can you please have a look?
I have separate data files for each scan, and I wanted to create pyart compatible file, having all scans in one.
You can download the data file here [archive.zip](https://drive.google.com/file/d/1jONSBlfhTPHRCFyEFKertzp7LYO7fzjr/view?usp=sharing)

zssherman, 2021-09-12:
Hi @syedhamidali

I couldn't reproduce the time error, however the error i did run into was in the cfradial write the sweep number is failing. The radar.nsweeps doesn't match the radar.sweep_numbers. To fix this I did this before the writer:

radar.nsweeps = 10

Also I saw the radar.gate_latitude['data'] etc lines were failing, this is due to the radar.latitude['data'] etc not being set with np.array, to fix this:

radar.latitude['data'] = np.array([ds.siteLat.values])
radar.longitude['data'] = np.array([ds.siteLon.values])
radar.altitude['data'] = np.array([ds.siteAlt.values])

I was then able to run and write the file out with no errors.

syedhamidali, 2021-09-12:
Dear @zssherman, Thank you.
Was it the right approach to compile the data into one?
Please let me know so that I could close this issue.

syedhamidali, 2021-09-13:
I also tried reading data with netCDF4, it fits data properly, but the time there is again wrong, as you can see in the figures.
[image]

[image]

[image]

kmuehlbauer, 2021-09-13:
@syedhamidali looks like you would need to use Linux timestamp seconds since 01.01.1970 :-)

syedhamidali, 2021-09-13:
Thanks, @kmuehlbauer - It solved the problem.

closed 2021-09-13 by syedhamidali