ARM-DOE/pyart #1012: Hello, AttributeError: 'numpy.float64' object has no attribute 'subs'

wshn25, 2021-09-23:
import numpy as np 
import sympy
def f(w):
    x,y=w
    return np.array([
                   x**2 + x*y - 10,
                   y + 3*x*(y**2) - 57,             
  ])
def Derfu(x,y):
    return 2*x + y
def Derfu(x,y):
    return x
def Derfv(x,y):
    return 3*y**2
def Derfv(x,y):
    return 6*x*y + 1

#Newton method
def Newton(w,max_iter):
    x,y = f(w)
    guesses=np.array([w])
    for n in range(0,max_iter):
        fx,fy = f(np.array([x,y]))
        Dux = Derfu(x,y).subs(x,1.5).subs(y,3.5)
        Duy = Derfu(x,y).subs(x,1.5).subs(y,3.5)
        Dvx = Derfv(x,y).subs(x,1.5).subs(y,3.5)
        Dvy = Derfv(x,y).subs(x,1.5).subs(y,3.5)
        #initial guess
        x0 = 1.5
        y0 = 3.5
        #value of u0 and v0
        u0 = (x**2 + x*y - 10).subs([(x,1.5), (y,3.5)])
        v0 = (y + 3*x*(y**2) - 57).subs([(x,1.5), (y,3.5)])
        #Determinant Jacobian
        a = sympy.sympify(['x**2 + x*y - 10', 'y + 3*x*(y**2) - 57'])
        J = sympy.zeros(len(a),len(vars))
        for i, fi in enumerate(a):
            for j, s in enumerate(vars):
                J[i,j] = sympy.diff(fi, s)
        J_new = J.subs(x,1.5).subs(y,3.5)
        Det_J = sympy.Matrix.det(J_matrix)
        #formula
        xn = x0 - ((u0*Dvy)-(v0*Duy)/Det_J)
        yn = y0 - ((v0*Duy)-(u0*Dvx)/Det_J)
        guesses=np.append(guesses,[[xn,yn]],axis=0)
   return guesses
 w = np.array([1,1], dtype=float)       
guesses = Newton(w, 5)
print("guesses : ")       
print(guesses) 

Can someone help me to solve this problem issue?
I have some difficulties on how to make a program of the non-linear equation by using Newton_Raphson Methode in Python, 
So, I have just checked all the possibilities and still encounter this issue.

And here's the problem:
AttributeError                            Traceback (most recent call last)
/tmp/ipykernel_46099/900935590.py in <module>
     87 w = np.array([1,1], dtype=float)
     88 
---> 89 guesses = Newton(w, 5)
     90 
     91 print("guesses : ")

/tmp/ipykernel_46099/900935590.py in Newton(w, max_iter)
     44         fx,fy = f(np.array([x,y]))
     45 
---> 46         Dux = Derfu(x,y).subs(x,1.5).subs(y,3.5)
     47 
     48         Duy = Derfu(x,y).subs(x,1.5).subs(y,3.5)

AttributeError: 'numpy.float64' object has no attribute 'subs'

THANKS.

zssherman, 2021-09-23:
@wshn25 I'm not sure how this relates to Py-ART, however, if your trying to substract which is my guess, its subtract not sub for numpy there is no np.sub in numpy. Closing this, unrelated.

closed 2021-09-23 by zssherman

wshn25, 2021-09-23:
@zssherman well, thank you. I got it.
The main aim of this script is just to find the roots from the non-linear equation. So, after running the program this issue was there and made me stuck. I also searched from many sources there is no np.sub in numpy.