ARM-DOE/pyart #1013: AttributeError: 'Radar' object has no attribute 'point_latitude'

wshn25, 2021-09-23:
**fig = plt.figure(figsize=[12, 6])
ax1 = fig.add_axes(map_panel_axes)
display = pyart.graph.GridMapDisplayBasemap(radar)
display.plot_basemap()
display_plot.grid('DBZH', level=level, vmin=vmin, vmax=vmax)
display.plot_crosshairs(lon=lon, lat=lat)**
---------------------------------------------------------------------------
I have some issues here. I am trying to find the answer in the Source code for pyart.graph.gridmapdisplay_basemap but still don't make it. Anyone who knows this, please kindly help to solve this issue. Thanks.**
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/tmp/ipykernel_37939/3423164769.py in <module>
      2 ax1 = fig.add_axes(map_panel_axes)
      3 display = pyart.graph.GridMapDisplayBasemap(radar)
----> 4 display.plot_basemap()
      5 display_plot.grid('DBZH', level=level, vmin=vmin, vmax=vmax)
      6 display.plot_crosshairs(lon=lon, lat=lat)

~/anaconda3/envs/py38/lib/python3.8/site-packages/pyart/graph/gridmapdisplay_basemap.py in plot_basemap(self, lat_lines, lon_lines, resolution, area_thresh, auto_range, min_lon, max_lon, min_lat, max_lat, ax, **kwargs)
     96         """
     97         # make basemap
---> 98         self._make_basemap(resolution, area_thresh, auto_range,
     99                            min_lon, max_lon, min_lat, max_lat, ax, **kwargs)
    100 

~/anaconda3/envs/py38/lib/python3.8/site-packages/pyart/graph/gridmapdisplay_basemap.py in _make_basemap(self, resolution, area_thresh, auto_range, min_lon, max_lon, min_lat, max_lat, ax, **kwargs)
    615         # determine map region
    616         if auto_range:
--> 617             max_lat = self.grid.point_latitude['data'][0].max()
    618             max_lon = self.grid.point_longitude['data'][0].max()
    619             min_lat = self.grid.point_latitude['data'][0].min()

AttributeError: 'Radar' object has no attribute 'point_latitude'

zssherman, 2021-09-23:
@wshn25 For gridmapdisplay your object needs to be a grid object not a radar object. I would convert to a grid using pyart.map.grid_from_radars(radar, etc) first.

wshn25, 2021-09-24:
@zssherman thank you for responding. I will try your suggestion.

closed 2021-12-16 by zssherman