ARM-DOE/pyart #1021: Aircraft navigation correction

aschueth, 2021-11-19:
In Solox, there is a correction that can be made to airborne radar data to correct for aircraft motion in order to get ground relative velocities. I can't for the life of me find the metadata to fix this manually, so I'm not sure where to start on implementing a pull request, but I think this would be a very valuable feature to have in pyart.

zssherman, 2021-11-19:
Hello @aschueth ! Just one question, is solox another package I assume? What could always happen, if so, it can be made an optional dependency. With implementing a PR, I think in corrections in pyart.corrections would work. Depending on how the code is I can try to help guide this more.

aschueth, 2021-11-19:
Whoops, sorry @zssherman, I meant solox as soloii/solo3, the old NCAR c++ radar editor. Again I have no idea how it is written or what metadata it looks at, but hopefully someone in this community might have some knowledge?

zssherman, 2021-11-19:
Ah gotcha, makes sense. Yeah, I can ask around to see. I'll also labels to this Issue tracker. I agree, I think this would be a valuable feature to have in pyart as well. Trying to think on how to approach this.

nguy, 2022-08-04:
It's been a long time since I looked at this code. Maybe @dstex knows how this is done?

FYI, the airborned specific variables are stored in the [radar object](https://github.com/ARM-DOE/pyart/blob/main/pyart/core/radar.py#L695-L709) and if memory serves me correctly, this either cfradial defined OR NOAA P-3 specific.

If it's a NOAA P-3 sigmet file, it looks like you might need to modify [this function](https://github.com/ARM-DOE/pyart/blob/f923551e1fd8de72c8db6c76e6bc9913bf635644/pyart/io/_sigmet_noaa_hh.py#L11) to include store one or all of the various ground speeds found in [the header](https://github.com/ARM-DOE/pyart/blob/f923551e1fd8de72c8db6c76e6bc9913bf635644/pyart/io/_sigmet_noaa_hh.py#L184). Again don't recall which ground speed order or anything. But hope this helps to start.

dstex, 2022-08-04:
This is something that is/was considered in the ongoing development of HawkEdit (the forthcoming solo replacement). They note the applicable corrections needed in these [meeting notes](https://github.com/NCAR/lrose-HawkEdit/blob/54b142e42eb3c01bc81df489a9ea0c4fab21ec06/meeting_notes/20220118.md). 

As far as I can tell, the aircraft motion variables are included in both the P-3 sigmet files and the cfradials. 

I'd be happy to look at working this into Py-ART in the near future (and/or testing any implementation against the solo-based correction).

aschueth, 2023-03-16:
Finally revisiting this issue and making some progress, but wanted to ask @zssherman among others how to actually apply these airborne platform velocities in the metadata. As of now, _sigmet_noaa_hh.py uses the filemetadata() function from config, and fills in that data appropriately. However, platform velocities are new and are not included in the metadata. Should I change filemetadata() in config to add the horizontal ground-relative platform velocity and vertical velocity metadata? If I do that, I need to not only fill that in _sigmet_noaa_hh, but also all of the the other possible ways to read in data which is... a lot. Is there a better way to add these velocity data in with _sigmet_noaa_hh.py without messing with the all of the io functions?

zssherman, 2023-03-17:
@aschueth I would have to take a look and see how to possibly implement it. For clarification, would you be adding a correction function that corrects for airborne motion?

zssherman, 2023-03-17:
I'll take a look at the sigment_noaa code as well, to maybe get a better understanding of it as well

aschueth, 2023-03-17:
Sorry, I should clarify, I believe there are two aspects to this. The first would be to add metadata to the radar object that describes the platform velocity (horizontal and vertical). My idea for this is to have it in the same place as the azimuth, elevation, tilt, roll variables, etc.. As far as I know, the _sigmet_noaa_hh io function would be the only one to add these data, and then the other io functions would just define these data as zero. Thinking more, I believe adding this metadata could be valuable outside of just aircraft platforms, but could be a placeholder to add truck-based mobile radar velocity, in case they scan while driving. I personally have had a situation where saving mobile, truck-based velocity metadata would be valuable.

The second aspect would be a completely independent function that would live in pyart.correct that would use the platform velocity metadata to correct the platform motion. I was considering correcting for the platform velocity automatically as long as it's not 0, but that solution might be unclear to the user on if the radial velocity is platform corrected or not. Therefore, having a separate function to correct for the platform velocity would be the most transparent.

These are just the ideas I have in my mind, but I welcome your thoughts and opinions, since adding these metadata in the radar object is relatively involved and I don't want to break anything.

aschueth, 2023-03-17:
As for physical progress, I have done relatively little, just verified that xhdr in _decode_noaa_hh_hdr() in _sigmet_noaa_hh.py was correctly reading the velocities saved in the sigmet file, and that I did not have to parse additional binary. Since the velocites are correct, I can create dictionaries from that data:
Line 113 ish:
```
     if position_source == "gps":
        lat_data = bin4_to_angle(xhdr["gps_lat"])
        lon_data = bin4_to_angle(xhdr["gps_long"])
        alt_data = xhdr["gps_alt"] / 100.0
        vel_ground_data = np.sqrt(xhdr["gps_vel_n"]**2+xhdr["gps_vel_e"]**2)
        vel_v_data = xhdr["gps_vel_v"]
    elif position_source == "aamps":
        lat_data = bin4_to_angle(xhdr["aamps_lat"])
        lon_data = bin4_to_angle(xhdr["aamps_long"])
        alt_data = xhdr["aamps_alt"] / 100.0
        vel_ground_data = xhdr["aamps_ground_vel"]
        vel_v_data = xhdr["aamps_vel_v"]
    elif position_source == "irs":
        lat_data = bin4_to_angle(xhdr["irs_lat"])
        lon_data = bin4_to_angle(xhdr["irs_long"])
        alt_data = xhdr["gps_alt"] / 100.0
        vel_ground_data = np.sqrt(xhdr["irs_vel_e"]**2+xhdr["irs_vel_n"]**2)
        vel_v_data = xhdr["irs_vel_v"]
    else:
        raise ValueError("Invalid position_source")

    latitude["data"] = lat_data
    longitude["data"] = lon_data
    altitude["data"] = alt_data

    vel_ground = {'units':'centimeters per second','standard_name':'vel_ground','long_name':'Ground Velocity'}
    vel_v = {'units':'centimeters per second','standard_name':'vel_v','long_name':'Vertical Velocity'}
    vel_ground['data'] = vel_ground_data
    vel_v['data'] = vel_v_data

```

But I need to save those dictionaries somewhere in the metadata

zssherman, 2023-03-17:
Makes sense! I'll dig further into the code and see the best way to add the metadata in.