ARM-DOE/pyart #1023: ODIM reader silently overwrites moment fields

kmuehlbauer, 2021-12-02:
The default moment mapping of the ODIM reader is

https://github.com/ARM-DOE/pyart/blob/e6a312bdb1bab09388952a72a5d634853022c42b/pyart/aux_io/odim_h5.py#L21-L38

If a file contains "DBZH" and "DBZV" then both get mapped to `reflectivity`:

https://github.com/ARM-DOE/pyart/blob/e6a312bdb1bab09388952a72a5d634853022c42b/pyart/aux_io/odim_h5.py#L108-L113

This will in consequence overwrite the first ("DBZH") by the second ("DBZV") moment (line 363 below) since `field_name` is `reflectivity`in both cases:

https://github.com/ARM-DOE/pyart/blob/e6a312bdb1bab09388952a72a5d634853022c42b/pyart/aux_io/odim_h5.py#L336-L363

This is also true for "TH/TV", "VRADH/VRADV" and possibly others. The major consequence is that users which have those polarimetric moments will only get one of these moments back. In the case of DBZH/DBZV they will get DBZV (vertical reflectivity) but might think they got DBZH (horizontal reflectivity).

I've not checked, but this might have to be fixed in other readers too.

Proposed solution: 

- Do not silently overwrite existing fields in the reader functions.
- If a field gets overwritten raise an informative error.

zssherman, 2021-12-14:
@kmuehlbauer Thanks for catching this! I'm check to see if this occurs in other readers as well. I think proposed solution 1 would work. I'm wondering why the original code, when it was added, didn't include separation with reflectivity_horizontal or reflectivity_vertical, If it was maybe just overlooked or something else. I'll look into this more, because one of these moments returned can't be correct.

kmuehlbauer, 2021-12-15:
@zssherman Does that mean we could just use 

`total_power_horizontal` / `total_power_vertical`
`reflectivity_horizontal` / `reflectivity_vertical`
`velocity_horizontal` / `velocity_vertical`
`spectrum_width_horizontal` / `spectrum_width_vertical`

and so on in the fieldnames mapping? Or is there more involved? Just asking because that looks like an easy fix.

zssherman, 2021-12-15:
@kmuehlbauer I'm hoping it should be an easy fix and just needs the renaming. I'm checking to see if those names are standard or should be fine to use. And if there is more to it possibly.

closed 2022-02-16 by zssherman