ARM-DOE/pyart #1028: Example data link is broken again

asiyats, 2022-01-02:
The link provided in README.md is no longer working, page not found, please update the new link

The files used in these examples are available for download_.

.. _download: https://engineering.arm.gov/~jhelmus/pyart_example_data/

zssherman, 2022-01-02:
Thanks @asiyats for the heads up, I'll take a look and go about finding a more permanent location.

zssherman, 2022-01-06:
Data link has been restored and the example data is accessible again. Leaving this issue open until we work on a more permanent location for the data.

mgrover1, 2022-03-10:
@zssherman can we close this now that we have the example hosted in a permanent location?

zssherman, 2022-03-10:
@mgrover1 One last thing, in the docs for the downloadable data the link is still https://engineering.arm.gov/~jhelmus/pyart_example_data/

There might be still one spot to change in the docs and we should be good.

zssherman, 2022-03-10:
The link seen here:
https://arm-doe.github.io/pyart/examples/index.html

PR #1068 (2022-03-10): Update examples data link to adc
Closes #1028 

This **should** be the last place in the docs that we need to change the link for
closed 2022-03-10 by mgrover1