ARM-DOE/pyart #1036: References for qpe related functions in pyart.retrieve

jaypotnis, 2022-02-23:
Hello,
I am using the qpe techniques under pyart.retrieve to estimate rain rate. I needed to research on the Quantitative Precipitation Estimation techniques used by pyart like est_rain_rate_kdp, est_rain_rate_z, est_rain_rate_z_poly, est_rain_rate_kdp.
Do these techniques have references to any published scientific papers?

zssherman, 2022-02-23:
Hi @jaypotnis, thanks for the heads up, I'm checking into this now. Seems to be added awhile back so trying to contact the original code authors on this.

zssherman, 2022-02-24:
@jaypotnis I reached out to the author.

For these rain rates:
The coefficients were taken from the paper https://rmets.onlinelibrary.wiley.com/doi/full/10.1002/qj.1934

As well as based on the report:
https://www.researchgate.net/publication/239575335_Operational_Use_of_Radar_for_Precipitation_Measurements_in_Switzerland

I will update the documentation as such to reflect this.

I can provide more of the discussion as I hear more!

zssherman, 2022-03-01:
@jaypotnis zpoly and z are based on:
Doelling, I. G., J. Joss, and J. Riedl, Systematic variations of Z-R relationships from drop size distributions measured in Northern Germany during seven years, Atmos. Res., 48, 635–649, 1998

as well as:
Joss, J., Sch¨adler, B., Galli, G.,
Cavalli, R., Boscacci, M.,
Held, E., Della Bruna, G.,
Kappenberger, G., Nespor, V.
and Spiess, R.
1998 Operational use of radar for precipitation measurements in
Switzerland. vdf Hochschulverlag AG, ETH Z¨urich, Switzerland

jaypotnis, 2022-03-08:
Thanks Zach. This helps a lot.

zssherman, 2022-03-11:
Added references to the code, closing for now in favor of #1042

closed 2022-03-11 by zssherman