ARM-DOE/pyart #1041: Determine Better Method of Dealing with Test Data

mgrover1, 2022-02-28:
Currently, the data are hosted on an ANL server, requiring users to download the data manually, move the data to the necessary directories, then build the docs. It would be useful to use a tool such as [Pooch](https://www.fatiando.org/pooch/latest/index.html) to help with this process, making it possible to build the example docs on Github Actions instead of running locally, then adding that to the docs.

mgrover1, 2022-02-28:
For now, I created a demo repo [here](https://github.com/mgrover1/pyart-data-demo) to test this out - once it is ready, Ill submit a PR

mgrover1, 2022-02-28:
Alrighty... [the prototype](https://github.com/mgrover1/pyart-data-demo) works 👍 with the syntax

```python
from pyart_datasets import get_test_data

radar = pyart.io.read_mdv(get_test_data('110635.mdv'))
```

This will make building the docs _much_ easier, especially the example gallery.

One of the remaining questions is whether this should be its own package (ex. `pyart_datasets`) or a module within PyART... 

@scollis @AdamTheisen @rcjackson @zssherman what are your thoughts here?

zssherman, 2022-03-01:
@mgrover1 Sweet, I think this will be much better. Think jjhelmus link is going to be removed. We're going to use https://adc.arm.gov/pyart/example_data/ instead, as a heads up. I think for the actually code itself is pooch a python native package or does it have to be installed as a dependency? Trying to think if it's the later how to go about that to avoid increasing the dependency count.

mgrover1, 2022-03-01:
@zssherman Sounds good! 

Pooch **does** need to installed as a dependency, but this is where we could think about whether we want this to be a _module_ within PyART, or create a separate package... I put some pros/cons to these approaches in [this afternoon's meeting notes](https://hackmd.io/eXMjluMdRMOEscaleNWYmQ)

dopplershift, 2022-03-01:
Just wanted to add that Pooch is pure Python and has been pretty stable. MetPy has had 0 problems depending on it.

zssherman, 2022-03-01:
@dopplershift Sweet sounds good! Yeah then I'm all for it being added. We can discuss more on the call today, but I have an idea of where to put it within pyart @mgrover1

PR #1048 (2022-03-02): Add example data into testing module
Closes #1041

Adds [Pooch](https://www.fatiando.org/pooch/latest/) method of accessing larger datasets, accessed by using the `pyart.testing` module. From a user's perspective, this would look something like:

```python
from pyart.io import read
from pyart.testing import get_test_data

data = get_test_data("110041.mdv")
radar = read(data)

...
```
closed 2022-03-02 by zssherman