ARM-DOE/pyart #1042: Improving references of methods in Py-ART

zssherman, 2022-03-01:
Mentioned by users as well as digging through the code from years ago. Looking into adding better reference to journals etc to methods. Working on qpe methods at the moment.

mgrover1, 2022-03-01:
Let's put a checklist together for this!

mgrover1, 2022-03-08:
@zssherman are there any remaining references that need to be added?

jrobrien91, 2022-03-08:
I can keep a list of missing/needed references as I go through PyART getting more familiar with the package.

mgrover1, 2022-03-08:
@jrobrien91 could you share that list here? That would be super helpful!!

zssherman, 2022-03-08:
@mgrover1 I'm not sure, alot of the older code I need to still go through. I opened this to kinda address them as they popped up as well.

mgrover1, 2022-11-22:
# Py-ART Reference Checklist

Here is a list of modules and associated references that are added or still need to be added

## pyart.retrieve
- [ ] calculate_snr_from_reflectivity

PR #1620 (2024-08-27): Add more references
Add additional references currently missing in the docs

- [x] Closes #1042 
- [x] Tests added
- [x] Documentation reflects changes
closed 2024-08-27 by mgrover1