ARM-DOE/pyart #1052: FIX: (ENH) Fix landing page for https://arm-doe.github.io/pyart/

scollis, 2022-03-04:
The very first image we see on https://arm-doe.github.io/pyart/ is using the Jet Colormap. We need to fix this

zssherman, 2022-03-04:
Ah yep good catch!

sobitp59, 2022-03-04:
@zssherman can you please elaborate what needs to be fixed

zssherman, 2022-03-04:
Hi @sobitp59, overtime we have moved all jet colormaps to the HomeyerRainbow colormap for reflectivity and more as it is Color Vision Deficiency friendly. Seems we missed one of our examples with the switch over.

mgrover1, 2022-03-04:
Good point @scollis - now that docs are building properly we can go ahead and make that change 😁

mgrover1, 2022-03-08:
Coming back to this - @sobitp59 I think the best steps forward would be the following:
- Fork this repo
- Go through one of the examples (for example, one from [plot ppi](https://arm-doe.github.io/pyart/examples/plotting/plot_ppi_mdv.html#sphx-glr-examples-plotting-plot-ppi-mdv-py))
- Save that image
- Reference that image in [index rst file](https://github.com/ARM-DOE/pyart/blob/main/doc/source/index.rst)

Currently, that page points to images in the [`_static` directory within the docs](https://github.com/ARM-DOE/pyart/tree/main/doc/source/_static), so be sure to name those images you create appropriately.

If you need any help, we are here to lend a hand where possible!

sobitp59, 2022-04-12:
@mgrover1 sorry I will not be able to do this, my exams are going on

closed 2022-06-23 by mgrover1