ARM-DOE/pyart #1078: pyart.io.read_cradial issue on a Mac (Big Sur 11.6.5)

wxwolff, 2022-03-18:
Hi,

I recently encounter an issue reading Cfradial files on my Mac. It used to work, but now gets a MaskError.  I can run on Linux with lots of warnings, but no errors.

Pyart: 1.11.8
Numpy: 1.21.2
Found 54 files to process.
File Type:  cf
Unzipping CF file...
Processing:  NPOL1_2018_0320_075650.cf
---------------------------------------------------------------------------
MaskError                                 Traceback (most recent call last)
/var/folders/vc/wrsvlv556kx14jvkf03s6p9c0000gn/T/ipykernel_36460/3570699089.py in <cell line: 110>()
    137             f = ungzip_file(cf)
    138             print("Processing: ",f)
--> 139             radar = pyart.io.read(f, file_field_names=True)
    140         else:
    141             f = cf

~/anaconda3/lib/python3.8/site-packages/pyart/io/auto_read.py in read(filename, use_rsl, **kwargs)
    110         else:
    111             dset.close()
--> 112             return read_cfradial(filename, **kwargs)    # CF/Radial
    113     if filetype == 'WSR88D':
    114         return read_nexrad_archive(filename, **kwargs)

~/anaconda3/lib/python3.8/site-packages/pyart/io/cfradial.py in read_cfradial(filename, field_names, additional_metadata, file_field_names, exclude_fields, include_fields, delay_field_loading, **kwargs)
    118     # 4.3 Global variable -> move to metadata dictionary
    119     if 'volume_number' in ncvars:
--> 120         metadata['volume_number'] = int(ncvars['volume_number'][:])
    121     else:
    122         metadata['volume_number'] = 0

~/anaconda3/lib/python3.8/site-packages/numpy/ma/core.py in __int__(self)
   4382                             "to Python scalars")
   4383         elif self._mask:
-> 4384             raise MaskError('Cannot convert masked element to a Python int.')
   4385         return int(self.item())
   4386 

MaskError: Cannot convert masked element to a Python int.

zssherman, 2022-03-18:
@wxwolff Seems the metadata is storing a mask value instead of possibly a string. I think I might have a work around for that. Do you have a file I can test on?

wxwolff, 2022-03-18:
Hi Zach,

Thanks for the really quick response. I do have a file, but it’s too big for email so I have posted it on our web site.  Try this link:

https://gpm-gv.gsfc.nasa.gov/Data/NPOL1_2022_0102_151855.cf.gz

Dave

_____________________________________
David B. Wolff, Chief Code 610.W
GPM Deputy Project Scientist for GV
NASA Wallops Flight Facility
34200 Fulton Street
Building N159, Room E220
Wallops Island, VA 23337
Ph: 757-824-1515; Cell: 443-262-1146

zssherman, 2022-03-18:
Ah you said mac is not working but linux is however. Gotcha so its probably os related

wxwolff, 2022-03-18:
> @wxwolff<https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fwxwolff&data=04%7C01%7CDavid.B.Wolff%40nasa.gov%7Cc954d9920ca9482cd89908da08f124fb%7C7005d45845be48ae8140d43da96dd17b%7C0%7C0%7C637832129045429191%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=2an0usLUdsUD26xu%2F%2BHupwxmEJ%2FIHGzSxA%2FJB9pAhm0%3D&reserved=0> Not a problem! And I haven't been able to reproduce the error yet. Are you adding metadata to the readers? Are you able to share the code your using that is causing this error?

Yes, it is a mac issue, but from what I can tell, at its root, it’s a numpy issue.

Davids-Mac-Pro.local> python --version
Python 3.8.12
Davids-Mac-Pro.local> python
Python 3.8.12 | packaged by conda-forge | (default, Jan 30 2022, 23:33:09)
[Clang 11.1.0 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyart

## You are using the Python ARM Radar Toolkit (Py-ART), an open source
## library for working with weather radar data. Py-ART is partly
## supported by the U.S. Department of Energy as part of the Atmospheric
## Radiation Measurement (ARM) Climate Research Facility, an Office of
## Science user facility.
##
## If you use this software to prepare a publication, please cite:
##
##     JJ Helmus and SM Collis, JORS 2016, doi: 10.5334/jors.119

>>> f = 'NPOL1_2018_0320_075650.cf'
>>> radar = pyart.io.read_cfradial(f)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/wolff/anaconda3/lib/python3.8/site-packages/pyart/io/cfradial.py", line 120, in read_cfradial
    metadata['volume_number'] = int(ncvars['volume_number'][:])
  File "/Users/wolff/anaconda3/lib/python3.8/site-packages/numpy/ma/core.py", line 4384, in __int__
    raise MaskError('Cannot convert masked element to a Python int.')
numpy.ma.core.MaskError: Cannot convert masked element to a Python int.
>>>

_____________________________________
David B. Wolff, Chief Code 610.W
GPM Deputy Project Scientist for GV
NASA Wallops Flight Facility
34200 Fulton Street
Building N159, Room E220
Wallops Island, VA 23337
Ph: 757-824-1515; Cell: 443-262-1146

zssherman, 2022-03-18:
@wxwolff Yeah, I'm confused on why it works for linux but not mac. I do have a possible fix which Ill be doing a PR here shortly. Are you locally installing pyart or is it conda installed?

wxwolff, 2022-03-18:
Conda.

_____________________________________
David B. Wolff, Chief Code 610.W
GPM Deputy Project Scientist for GV
NASA Wallops Flight Facility
34200 Fulton Street
Building N159, Room E220
Wallops Island, VA 23337
Ph: 757-824-1515; Cell: 443-262-1146

zssherman, 2022-03-18:
Trying to find a way to test this real quick, sadly i'm on a linux system. But in theory the PR #1080 should fix it as it will flatten and retrieve the volume number before using int on it.

mgrover1, 2022-03-18:
@wxwolff which package/function is this? What are you using for unzipping in python?

```python
137 f = ungzip_file(cf)
```

mgrover1, 2022-03-18:
After unzipping the file from the command line, I was able to read in the file with the following setup:
* macOS Monterey 12.2.1
* Numpy: 1.21.2

wxwolff, 2022-03-18:
#1080 resolved my issue!  Thanks Zach!!

closed 2022-03-18 by wxwolff