ARM-DOE/pyart #1114: Potential Correction in NEXRAD Level3 DPR  product reading

jaypotnis, 2022-04-16:
I am doing project on Quantitative Precipitation Estimation using WSR 88D Radar. I am comparing NEXRAD L3 Digital Precipitation Rate with actual Rain Rate from TRMM GV Rain Gauges.

Looks like the data being read in the below snippet does not scale the rain rate data correctly.
https://github.com/ARM-DOE/pyart/blob/a68bed26f6380eecbf1042c94bd4da0b60fb2da7/pyart/io/nexrad_level3.py#L297-L300

The way I am seeing the data, I need to multiply the whole data by 10. To convert it to the proper scale (i.e. inch/mm).

Below are some Z-RR plots at a constant height. Z is obtained from NEXRAD L2 data. Additionally I am also multiplying the Pyart read DPR data by 25.4 to convert it to mm/hr. But that is not expected by pyart as L3 DPR data is supposed to have default unit as inch/mm.

L3 DPR data after reading it through pyart and converting to mm/hr.
[image]

L3 DPR data after reading it through pyart, multiplying by 10 and converting to mm/hr.
[image]

I hope this explanation helped. This is not a significant issue but would still be good to avoid confusion while reading the DPR data.

Btw, thanks for creating this wonderful tool for reading L3 data. There are not a lot of tools that do that. Found it very helpful. :)

mgrover1, 2022-04-18:
Thank you @jaypotnis for pointing this out. We will look into implementing a solution to this, or you are welcome to submit a pull request (PR) with the fix if you are interested. Happy to help with this process!

dopplershift, 2022-04-18:
I'm confused, why isn't it expected? In the XDR product, the unit are encoded:
```
Unit = inches/hour
```
So it seems to me that needing to multiply by 25.4 to get mm/hr is correct?

mgrover1, 2022-04-18:
The units pyart returns are wrong - it is supposed to inches/hour.

Also, I think I found the issue... we are assuming the data is an 8 bit integer

https://github.com/ARM-DOE/pyart/blob/main/pyart/io/nexrad_level3.py#L219

but according to the Level3 documentation, this field is a 16 bit integer 
[image]

Currently working on a PR to fix both of these issues, and add some tests...

mgrover1, 2022-04-18:
Alrighty - I tested this out on one of the TRACER cases (a field campaign coordinated by ARM), which includes disdrometers/other ground precipitation measurements. 

The laser disdrometer (ld), located near the airport, measured a maximum precipitation rate of 95 mm/hr, which corresponds to 3.7 inches/hour. 

[image]

NEXRAD data shows rain rates similar, with a few values **much** higher (>7 inches/hour)... here is the rain rate (`DPR`) field at the time closest to that peak disdrometer data:
[image]

mgrover1, 2022-04-18:
Keeping this open for now - here is the figure from the previous method (before the merged PR)
[image]

@dopplershift / @jaypotnis does this look like it's solved this issue?

dopplershift, 2022-04-19:
It certainly looks more sensible to me.

And no prob @mgrover1, taking a look at this has led me to fix a different bug in MetPy's own support for this product.

closed 2022-04-19 by mgrover1

jaypotnis, 2022-04-19:
The rain rate certainly looks better now. Thanks for a quick resolution on this @mgrover1

PR #1116 (2022-04-18): FIX: Fix IO for NEXRAD Level 3 instantaneous precipitation rate (DPR)
Fixes the incorrect units and incorrect dealing of 16 bit integers from the NEXRAD Level 3 DPR data

Closes #1114