ARM-DOE/pyart #1161: importing pyart causes "DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead."

chudlerk, 2022-05-25:
Whenever I `import pyart` I get the following slew of warnings:

```
C:\Users\kchudler\Miniconda3\lib\site-packages\xarray\core\pycompat.py:22: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  duck_array_version = LooseVersion(duck_array_module.__version__)
C:\Users\kchudler\Miniconda3\lib\site-packages\xarray\core\pycompat.py:37: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  duck_array_version = LooseVersion("0.0.0")
C:\Users\kchudler\Miniconda3\lib\site-packages\xarray\core\pycompat.py:37: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  duck_array_version = LooseVersion("0.0.0")
C:\Users\kchudler\Miniconda3\lib\site-packages\setuptools\_distutils\version.py:351: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  other = LooseVersion(other)
C:\Users\kchudler\Miniconda3\lib\site-packages\setuptools\_distutils\version.py:351: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  other = LooseVersion(other)
C:\Users\kchudler\Miniconda3\lib\site-packages\xarray\core\npcompat.py:82: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(np.__version__) >= "1.20.0":
C:\Users\kchudler\Miniconda3\lib\site-packages\setuptools\_distutils\version.py:351: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  other = LooseVersion(other)
C:\Users\kchudler\Miniconda3\lib\site-packages\xarray\core\pdcompat.py:45: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(pd.__version__) < "0.25.0":
C:\Users\kchudler\Miniconda3\lib\site-packages\setuptools\_distutils\version.py:351: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  other = LooseVersion(other)
C:\Users\kchudler\Miniconda3\lib\site-packages\paramiko\transport.py:219: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,
```

I am running the latest version of pyart. I am using Anaconda on a Windows machine for my package managing. I have tried removing and reinstalling the pyart package.

Thanks!

Let me know if you need any more information of if this issue should be posted elsewhere.

zssherman, 2022-05-25:
Thanks for the heads up @chudlerk , the warnings are interesting, what are your versions for setuptools and xarray? Trying to narrow down the possible cause. Wondering too, if how we are pinning numpy if maybe an older distutils is being pulled as well.

chudlerk, 2022-05-25:
xarray: 0.20.1
setuptools: 61.2.0

zssherman, 2022-05-25:
Gotcha, what python version are you using? From what i've looked up, seems in python 3.10 disutils is phased out so any dependencies that still use it will throw the warning, might be a possible cause. So possibly might be xarray might need to update. Another option is xarray is at 0.21 and setuptools 62.3.2, so might be a depreciation warning there as well. I'll keep digging, see if I can find a solution.

chudlerk, 2022-05-25:
I saw that too, but I'm only using Python 3.9.12 so I'm not sure if that's the issue

zssherman, 2022-05-25:
Ah gotcha.... hmm... yeah i'll keep digging and see what I can find.

mgrover1, 2022-05-31:
@chudlerk it looks like this issue was resolved in the January release (v0.21.1) of Xarray, indicated on their changelog (https://docs.xarray.dev/en/stable/whats-new.html#v0-21-1-31-january-2022). Can you try updating Xarray to at least 0.21.1 and see if that solves the issue?

chudlerk, 2022-06-01:
^That solved the issue. Had to update through the conga-forge channel to get the latest version installed. I also had to update paramiko to get rid of one of the warnings.

mgrover1, 2022-06-01:
Great! We will need to pin to at least that xarray version. Thanks or asking!

PR #1167 (2022-06-07): Pin version of xarray
Closes #1161
closed 2022-06-07 by mgrover1