ARM-DOE/pyart #1172: Old NEXRAD "missing" lat lon

dopplerchase, 2022-06-11:
I was playing around with some old NEXRAD data (Sept 27 1998 from KBUF; 4 inch hail day) and I noticed that these old files have their latitude and longitude a bit messed up. It seems to be centered on the radar location 

[image]

Reproduce the issue here: 

https://colab.research.google.com/drive/1PeFXuAez7P7pXbkYixTys0pawevSuyKm?usp=sharing

I think its a decently easy fix if you can parse the name of the file and then get the lat,lon of the radar location. 

Thats what i did as a work around:
```
    rad_lon=-78.73667
    rad_lat=42.94889
    lon = rad_lon + radar.gate_longitude['data'][s_idx:e_idx,:]
    lat = rad_lat +radar.gate_latitude['data'][s_idx:e_idx,:]
```

has this already been implemented somewhere in pyart? if not, where do you see it being implemented? (so I can maybe start drafting a function to help fix)

mgrover1, 2022-06-11:
Great question @dopplerchase - we currently do not make any assumptions about reading the lat/lon from the filename, but you can pass the station id into the [`pyart.io.read_nexrad_archive`](https://github.com/ARM-DOE/pyart/blob/main/pyart/io/nexrad_archive.py#L19) function, as `station="KBUF"`. We have functions that are able to index the name of the radar, and use the lats and lons of the radar site. 

As is the case with the older data, if it can't find the radar lat and lon, it will set it to 0, which is what happened here.

mgrover1, 2022-06-11:
The IO line would look something like this:

```python
import pyart 
radar = pyart.io.read('./KBUF19980927_204203.gz', station="KBUF")
```

We could consider trying to extract the path from the filename, checking to see if it is in our database of sites:

```python
path = pathlib.Path(filepath)
radar_site = path.stem[:4]

if radar_site in sites.keys():
    radar_location = get_lat_lon(radar_site)
else:
   radar_location = (0 , 0, 0)
```

Or something like that...