ARM-DOE/pyart #1186: Compressed data ended before the end-of-stream marker was reached

wxbinge, 2022-06-23:
This error is easily resolved by redownloading the radar data, but I'm running this particular script operationally, so when this error occurs, the script exits and I would miss the cycle. 

I was wondering is there a way to bypass this error?

I am pulling the radar data from the TGFTP server.

Traceback (most recent call last):
  File "/wfs/radar/pyscripts/national_composite_archival.py", line 186, in <module>
    radar.append(pyart.io.read_nexrad_level3(i))				# Opening radar file and appending each file to an array
  File "/anaconda3/envs/WFS/lib/python3.10/site-packages/pyart/io/nexradl3_read.py", line 78, in read_nexrad_level3
    nfile = NEXRADLevel3File(prepare_for_read(filename, storage_options=storage_options))
  File "/anaconda3/envs/WFS/lib/python3.10/site-packages/pyart/io/nexrad_level3.py", line 146, in __init__
    buf2 = bz2.decompress(buf[bpos:])
  File "/anaconda3/envs/WFS/lib/python3.10/bz2.py", line 341, in decompress
    raise ValueError("Compressed data ended before the "
**ValueError: Compressed data ended before the end-of-stream marker was reached**

zssherman, 2022-06-23:
@wxbinge Hmm would maybe a try and except, catch the error and rerun the download on the file or are you using pyarts nexrad reader to download the file?

wxbinge, 2022-06-24:
> 

I don't know why 'try and except' didn't cross my mind, but you're right. That will work to bypass the error. I'm using bash to download the radar files, so I'm going to need to do something creative to see if I can invoke a download from within the python script if the error arises.

mgrover1, 2022-06-24:
@wxbinge that error occurs when the file is not completely downloaded. 

See the stack overflow discussion [here](https://stackoverflow.com/questions/40877781/eoferror-compressed-file-ended-before-the-end-of-stream-marker-was-reached-mn)

You will need to add logic within your script to check to see if the file is completely downloaded yet, or consider streaming the data from Amazon Web Services (AWS), as we did in this example:
-  [NEXRAD AWS Example](https://arm.doe.github.io/pyart/examples/io/plot_nexrad_data_aws.html#sphx-glr-examples-io-plot-nexrad-data-aws-py])