ARM-DOE/pyart #1202: No normalized coherent power in my radar field

raffyjef, 2022-07-14:
Hi, I just want to ask if how can I use the functions with ncp if my radar file do not have it in its field. I can use the gatefilter function but not the others. My radar file is in .uf format. Thank you

zssherman, 2022-07-14:
Hi @raffyjef, what other functions are you trying to use? Talking to Scott Collis, you could make a ncp field of ones to get you going. To do that, can do:
```
ncp_values = np.ones((radar.nrays, radar.ngates))
ncp = pyart.config.get_metadata('normalized_coherent_power')
ncp['data'] = ncp_values
radar.add_field('normalized_coherent_power', ncp)

raffyjef, 2022-07-15:
I am trying to perform attenuation correction using pyart.correct.calculate_attenuation. It did work. Thank you for this!

closed 2022-07-15 by raffyjef