ARM-DOE/pyart #1217: PyArt does not display any map when using "RadarDisplay"

softieeeeee, 2022-07-23:
So I'm following the [tutorial](https://nexradaws.readthedocs.io/en/latest/Tutorial.html) and when I get to this part and run it (along with the other code):
[image]
No map is generated or anything (there is no errors either, and I have all the dependencies).

What I expect to get:
[image]

But upon running it (in command prompt), I get zero results. No image file generated or window pop up of the radar data.

zssherman, 2022-07-24:
@softieeeeee Hmmm I know notebooks can be tricky with plots showing. Can you try plt.show() at the end of your code or
```
%matplotlib inline
````
in your import cell.

softieeeeee, 2022-07-25:
Using `%matplotlib inline` just errors out with 'invalid syntax'.
[image]

And, there is no such function named 'show()'.
[image]

zssherman, 2022-07-25:
Can't tell, is that calling plt.show()? I would try maybe
```
%matplotlib notebook

What system are you using, I am using linux and plt.show() works, but not sure if it changes between OS

softieeeeee, 2022-07-25:
I'm on Windows 10 64-bit (editing the file in VSCode). Maybe that is likely the reason it doesn't work.

mgrover1, 2022-07-25:
@softieeeeee - can you try printing `results`?

```python
results = conn.download(scans[0:4], templocation)
print(results)
```

That way we can ensure that the files are downloaded, and stored locally.

If that works, then make sure to add the figure to the plotting line:

```python
fig = plt.figure(figsize=(16,12))
for i,scan in enumerate(results.iter_success(),start=1):
    ax = fig.add_subplot(2,2,i)
    radar = scan.open_pyart()
    display = pyart.graph.RadarDisplay(radar)
    display.plot('velocity', 1, ax=ax, fig=fig, title="{} {}".format(scan.radar_id,scan.scan_time))
    display.set_limits((-150, 150), (-150, 150), ax=ax)
```

Give that a try and let us know if that solves this issue.