ARM-DOE/pyart #1229: Question regarding embellish key in plot_grid

wolfidan, 2022-08-03:
Dear Py-ART team,

In the [plot_grid](https://github.com/ARM-DOE/pyart/blob/main/pyart/graph/gridmapdisplay.py) function, there is an embellish key. 

It's specified that 

embellish : bool
            True by default. Set to False to supress drawinf of coastlines
            etc... Use for speedup when specifying shapefiles.
            Note that lat lon labels only work with certain projections.

When setting embellish=False, no country borders and coastlines are displayed which is fine and perfectly agrees with description, but the coordinates on the plot and the gridlines are not displayed either, which is a bit problematic (at least for me). I don't know if that was intended. Setting embellish=False is useful for the European users because the default geodata plotted by Py-ART are not so relevant for us, so it's easier for us to do it a-posteriori outside of Py-ART. 

A very simple fix for that would be to unindent l.249 to 276 to put them outside of the embellish condition. 
So embellish = False would mean no additional geodata is displayed but the plot still has  gridlines and x and y ticks.

I would be happy to hear your opinion on this point..
Cheers,
Daniel W.

zssherman, 2022-08-03:
Hmmm, I'm wondering too if they can be separated, embellish for coastlines etc, gridlines for the ticks etc. Having the gridlines included seems useful, but if there is any reason to not include, might be better to have a switch for them.

wolfidan, 2022-08-04:
Thanks @zssherman, we could also imagine having two switches, one could be _embellish_ to specify whether to plot the additional coastlines, countries, etc and could be _display_georef_ (for example) which would trigger the display of the gridlines and the coordinates.

zssherman, 2023-05-18:
Addressed in #1297 closing.