ARM-DOE/pyart #1245: plot_ppi_map with lat/lon

srbrodzik, 2022-08-17:
I have some code that I've been using successfully on a Debian 10 machine running pyart v1.11.2.  I've tried running it on a Debian 11 machine running pyart v1.12.5 and I get plots but no lat/lon information.  See my code and two attached images.  Can you explain what I might be missing?  

`import os
import sys
import matplotlib.pyplot as plt
import pyart
import numpy as np
from datetime import datetime
import cartopy.crs as ccrs
import warnings
warnings.filterwarnings("ignore") 

fname = "/home/disk/monsoon/precip/cfradial/spol_ncar/cfradial/rate/sur/20220607/cfrad.20220607_042450.162_to_20220607_043052.364_SPOL_SUR.nc"
outdir_base = '/home/disk/monsoon/precip/radar/spol_test'

# get date and time from filename
filename = os.path.basename(fname)
(prefix,start,junk,junk,junk) = filename.split('.')
start_obj = datetime. strptime(start, '%Y%m%d_%H%M%S')
start_str = start_obj.strftime('%Y/%m/%d %H:%M:%S')
datetime_str = start_obj.strftime('%Y%m%d%H%M%S')
date = start_obj.strftime('%Y%m%d')

# create outdir if necessary
outdir = outdir_base+'/'+date
if not os.path.exists(outdir):
    os.makedirs(outdir)

# read input file & get elevation angles
radar = pyart.io.read(fname)
radar.scan_type = 'sur'
els = radar.fixed_angle['data']

# get lowest elevation angle
index = 0
angle = round(els[index],1)
print('lowest angle =',angle)

# check to see if image already exists
file_out = 'research.Radar_SPOL.'+datetime_str+'.ppim_rrate_'+str(angle).replace('.','')+'.png'
if not os.path.isfile(outdir+'/'+file_out):

    # remove transition angle flags
    subset = radar.extract_sweeps([index])
    trans = subset.antenna_transition["data"]
    trans[:] = 0
    subset.antenna_transition["data"] = trans

    # create display
    # use 'subset' instead of 'radar' to define display
    # then sweep will be 0 for all plots since subset contains only one sweep
    display = pyart.graph.RadarMapDisplay(subset)
    fig = plt.figure(figsize=(12, 4.5))
    fig.tight_layout()
    fig.suptitle('SPOL Rain Rates '+start_str+' UTC PPI '+str(angle)+' deg', fontsize=18)

    ax = fig.add_subplot(121, projection=ccrs.PlateCarree())
    display.plot_ppi_map('RATE_HYBRID', sweep=0, ax=ax, title='',
                         vmin=0,vmax=50,
                         width=400000, height=400000,
                         colorbar_label='RATE_HYBRID (mm/hr)',
                         cmap='pyart_HomeyerRainbow',
                         lat_lines = np.arange(22,27,.5),
                         lon_lines = np.arange(118, 123,1),
                         #axislabels=('', 'Distance from radar (km)'),
                         resolution = '10m')
    
    ax = fig.add_subplot(122, projection=ccrs.PlateCarree())
    display.plot_ppi_map('RATE_PID', sweep=0, ax=ax, title='',
                         vmin=0,vmax=50,
                         width=400000, height=400000,
                         colorbar_label='RATE_PID (mm/hr)',
                         #cmap='pyart_Carbone42',
                         cmap='pyart_HomeyerRainbow',
                         lat_lines = np.arange(22,27,.5),
                         lon_lines = np.arange(118, 123,1),
                         #axislabels=('', 'Distance from radar (km)'),
                         resolution = '10m')

    # save plot
    plt.savefig(outdir+'/'+file_out)
`
[image]
[image]

mgrover1, 2022-08-18:
@srbrodzik can you add `embellish = True` to see if that solves the issue?

srbrodzik, 2022-08-18:
I'm not familiar with that command or how to use it and couldn't find anything about it on the web, so I put it after the display and fig command and before the individual plots are defined, like this:    

    display = pyart.graph.RadarMapDisplay(subset)
    fig = plt.figure(figsize=(12, 4.5))
    fig.tight_layout()
    fig.suptitle('SPOL Rain Rates '+start_str+' UTC PPI '+str(angle)+' deg', fontsize=18)
    embellish = True
 
It made no difference.  

If there's a more proper way to use the command, let me know.

mgrover1, 2022-08-18:
Could you add it in the plot_ppi_map? Ex. Plot_ppi_map(embellish=True)

mgrover1, 2022-08-19:
```python
# remove transition angle flags
subset = radar.extract_sweeps([index])
trans = subset.antenna_transition["data"]
trans[:] = 0
subset.antenna_transition["data"] = trans

# create display
# use 'subset' instead of 'radar' to define display
# then sweep will be 0 for all plots since subset contains only one sweep
display = pyart.graph.RadarMapDisplay(subset)
fig = plt.figure(figsize=(12, 4.5))
fig.tight_layout()
fig.suptitle('SPOL Rain Rates '+start_str+' UTC PPI '+str(angle)+' deg', fontsize=18)

ax = fig.add_subplot(121, projection=ccrs.PlateCarree())
display.plot_ppi_map('RATE_HYBRID', sweep=0, ax=ax, title='',
                     vmin=0,vmax=50,
                     width=400000, height=400000,
                     colorbar_label='RATE_HYBRID (mm/hr)',
                     cmap='pyart_HomeyerRainbow',
                     lat_lines = np.arange(22,27,.5),
                     lon_lines = np.arange(118, 123,1),
                     #axislabels=('', 'Distance from radar (km)'),
                     embellish=True,
                     resolution = '10m')

ax = fig.add_subplot(122, projection=ccrs.PlateCarree())
display.plot_ppi_map('RATE_PID', sweep=0, ax=ax, title='',
                     vmin=0,vmax=50,
                     width=400000, height=400000,
                     colorbar_label='RATE_PID (mm/hr)',
                     #cmap='pyart_Carbone42',
                     cmap='pyart_HomeyerRainbow',
                     lat_lines = np.arange(22,27,.5),
                     lon_lines = np.arange(118, 123,1),
                     #axislabels=('', 'Distance from radar (km)'),
                     embellish=True,
                     resolution = '10m')

# save plot
plt.savefig(outdir+'/'+file_out)
```

srbrodzik, 2022-08-19:
That doesn't result in any change.

I've attached my input file for you.
 cfrad.20220607_042450.162_to_20220607_043052.36...
<https://drive.google.com/file/d/1akTKyeLQT4npB68vJGLa_5I0Xg_0veyO/view?usp=drive_web>
Stacy

mgrover1, 2022-08-20:
I tried troubleshooting with version 1.12.5, and was not able to reproduce the error 

```python
# remove transition angle flags
subset = radar.extract_sweeps([index])
trans = subset.antenna_transition["data"]
trans[:] = 0
subset.antenna_transition["data"] = trans

# create display
# use 'subset' instead of 'radar' to define display
# then sweep will be 0 for all plots since subset contains only one sweep
display = pyart.graph.RadarMapDisplay(subset)
fig = plt.figure(figsize=(12, 4.5))
fig.tight_layout()
fig.suptitle('SPOL Rain Rates '+start_str+' UTC PPI '+str(angle)+' deg', fontsize=18)

ax = fig.add_subplot(121, projection=ccrs.PlateCarree())
display.plot_ppi_map('RATE_HYBRID', sweep=0, ax=ax, title='',
                     vmin=0,vmax=50,
                     width=400000, height=400000,
                     colorbar_label='RATE_HYBRID (mm/hr)',
                     cmap='pyart_HomeyerRainbow',
                     lat_lines = np.arange(22,27,.5),
                     lon_lines = np.arange(118, 123,1),
                     #axislabels=('', 'Distance from radar (km)'),
                     # embellish=True,
                     resolution = '10m')

ax = fig.add_subplot(122, projection=ccrs.PlateCarree())
display.plot_ppi_map('RATE_PID', sweep=0, ax=ax, title='',
                     vmin=0,vmax=50,
                     width=400000, height=400000,
                     colorbar_label='RATE_PID (mm/hr)',
                     #cmap='pyart_Carbone42',
                     cmap='pyart_HomeyerRainbow',
                     lat_lines = np.arange(22,27,.5),
                     lon_lines = np.arange(118, 123,1),
                     #axislabels=('', 'Distance from radar (km)'),
                     # embellish=True,
                     resolution = '10m')

# save plot
plt.savefig(outdir+'/'+file_out)
```
[image]

There is one way to override the grid labels, by adding 

```python
ax.gridlines(xlocs=lon_lines, ylocs=lat_lines, draw_labels=True)
```

When you create the plots. I added a revised code snippet similar to above, except with the added line to make sure that grid lines are added and labels are included! I hope this helps!

```python
index = 0 

# remove transition angle flags
subset = radar.extract_sweeps([index])
trans = subset.antenna_transition["data"]
trans[:] = 0
subset.antenna_transition["data"] = trans

# create display
# use 'subset' instead of 'radar' to define display
# then sweep will be 0 for all plots since subset contains only one sweep
display = pyart.graph.RadarMapDisplay(subset)
fig = plt.figure(figsize=(12, 4.5))
fig.tight_layout()
fig.suptitle('SPOL Rain Rates '+start_str+' UTC PPI '+str(angle)+' deg', fontsize=18)
lat_lines = np.arange(22,27,.5)
lon_lines = np.arange(118, 123,1)

ax = fig.add_subplot(121, projection=ccrs.PlateCarree())
display.plot_ppi_map('RATE_HYBRID', sweep=0, ax=ax, title='',
                     fig=None,
                     vmin=0,vmax=50,
                     width=400000, height=400000,
                     colorbar_label='RATE_HYBRID (mm/hr)',
                     cmap='pyart_HomeyerRainbow',
                     lat_lines = lat_lines,
                     lon_lines = lon_lines,
                     #axislabels=('', 'Distance from radar (km)'),
                     embellish=True,
                     resolution = '10m')

# Add grid labels to the first plot
gl = ax.gridlines(xlocs=lon_lines, ylocs=lat_lines, draw_labels=True)
gl.top_labels = False
gl.right_labels = False

ax2 = fig.add_subplot(122, projection=ccrs.PlateCarree())
display.plot_ppi_map('RATE_PID', sweep=0, ax=ax2, title='',
                     fig=None,
                     vmin=0,vmax=50,
                     width=400000, height=400000,
                     colorbar_label='RATE_PID (mm/hr)',
                     #cmap='pyart_Carbone42',
                     cmap='pyart_HomeyerRainbow',
                     lat_lines = np.arange(22,27,.5),
                     lon_lines = np.arange(118, 123,1),
                     #axislabels=('', 'Distance from radar (km)'),
                     embellish=True,
                     resolution = '10m')

# Add grid labels to the other plot
gl2 = ax2.gridlines(xlocs=lon_lines, ylocs=lat_lines, draw_labels=True)
gl2.top_labels = False
gl2.right_labels = False
```

srbrodzik, 2022-08-22:
Thanks for the feedback Max.  Unfortunately, that didn't fix my problem.
Could it be that a newer version of another package is required?

Stacy

mgrover1, 2022-08-23:
@srbrodzik can you run `conda list` and post the output here?

srbrodzik, 2022-08-23:
I'm not using anaconda.  I'm working on a Debian 11 system and have
attached a 'dpkg -l' listing.  PyArt is not listed due to how github
downloads interact with our package management system.  But I can assure
you that the version is as I've told you.

I'm starting to think this is a matplotlib problem and not a pyart
problem.  I had similar problems with axis labelling in some skewt plotting
code that worked with older versions of matplotlib.  I learned that
matplotlib was handling axes differently but never quite figured out how to
use it.  That might also explain why your 'fix' didn't work.

Stacy

mgrover1, 2022-08-23:
Ahh okay - I am not seeing the dpkg list...

Can you try updating matplotlib to a recent version? That might fix it or is at least worth a try.

srbrodzik, 2022-08-23:
Already using this:

ii  python3-matplotlib                            3.3.4-1
                 amd64        Python based plotting system in a style
similar to Matlab (Python 3)

Will try 3.5.3 and see what happens.

Stacy

mgrover1, 2022-08-24:
Okay! Let me know how that goes... fingers crossed that fixes it!

srbrodzik, 2022-08-29:
We installed the latest version of matplotlib from the git site which dpkg
-l identifies as v3.6.0:

ii  python3-matplotlib
                           3.6.0~dev3106+g07af522369-1                amd64
       Python plotting package

Not sure how that happened since I thought 3.5.3 was the most recent
version.

Bottom line, it makes things worse.  Now I don't even get plots without
lat/lon markings but instead get a bunch of error messages and no plots.
There's few references to cartopy so I checked the version we're using:

ii  python-cartopy-data                           0.20.2+dfsg-2
                             all          cartographic library for Python
(package data)
ii  python3-cartopy                               0.20.2+dfsg-2
                             amd64        Cartographic library for Python 3

Here's the errors:

In [6]:    ax = fig.add_subplot(121, projection=ccrs.PlateCarree())
  ...:     display.plot_ppi_map('RATE_HYBRID', sweep=0, ax=ax, title='',
  ...:                          fig=None,
  ...:                          vmin=0,vmax=50,
  ...:                          width=400000, height=400000,
  ...:                          colorbar_label='RATE_HYBRID (mm/hr)',
  ...:                          cmap='pyart_HomeyerRainbow',
  ...:                          lat_lines = lat_lines,
  ...:                          lon_lines = lon_lines,
  ...:                          #axislabels=('', 'Distance from radar
(km)'),
  ...:                          embellish = True,
  ...:                          resolution = '10m')
  ...:
  ...:
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-6-529f5263cc05> in <module>
     1 ax = fig.add_subplot(121, projection=ccrs.PlateCarree())
----> 2 display.plot_ppi_map('RATE_HYBRID', sweep=0, ax=ax, title='',
     3                      fig=None,

     4                      vmin=0,vmax=50,
     5                      width=400000, height=400000,

/usr/lib/python3/dist-packages/pyart/graph/radarmapdisplay.py in
plot_ppi_map(self, field, sweep, mask_tuple, vmin, vmax, cmap, norm,
mask_outside, title, title_flag, colorbar_
flag, colorbar_label, ax, fig, lat_lines, lon_lines, projection, min_lon,
max_lon, min_lat, max_lat, width, height, lon_0, lat_0, resolution,
shapefile, shapefile_kwargs, edges
, gatefilter, filter_transitions, embellish, raster, ticks, ticklabs,
alpha, edgecolors, **kwargs)

   298         if norm is not None:  # if norm is set do not override with
vmin/vmax
   299             vmin = vmax = None
--> 300         pm = ax.pcolormesh(x * 1000., y * 1000., data, alpha=alpha,
   301                            vmin=vmin, vmax=vmax, cmap=cmap,

   302                            edgecolors=edgecolors, norm=norm,

/usr/lib/python3/dist-packages/cartopy/mpl/geoaxes.py in wrapper(self,
*args, **kwargs)
   316
   317         kwargs['transform'] = transform
--> 318         return func(self, *args, **kwargs)
   319     return wrapper
   320

/usr/lib/python3/dist-packages/cartopy/mpl/geoaxes.py in pcolormesh(self,
*args, **kwargs)
  1802         result.__class__ = cartopy.mpl.geocollection.GeoQuadMesh
  1803
-> 1804         self.autoscale_view()
  1805         return result
  1806

/usr/lib/python3/dist-packages/cartopy/mpl/geoaxes.py in autoscale_view(self,
tight, scalex, scaley)
   944                                             scalex=scalex,
scaley=scaley)
   945         # Limit the resulting bounds to valid area.
--> 946         if scalex and self._autoscaleXon:
   947             bounds = self.get_xbound()
   948             self.set_xbound(max(bounds[0],
self.projection.x_limits[0]),

AttributeError: 'GeoAxesSubplot' object has no attribute '_autoscaleXon'

kmuehlbauer, 2022-08-30:
Welcome to dependency hell, @srbrodzik. This was fixed in cartopy here: https://github.com/SciTools/cartopy/pull/2054

But I do not know if this is already in their latest release.

kmuehlbauer, 2022-08-30:
Ok, looks like we have to wait for the next cartopy release: https://github.com/SciTools/cartopy/pull/2054#issuecomment-1221418391

mgrover1, 2022-08-30:
Yeah - @srbrodzik for some reason your environment has the new experimental version of matplotlib, which causes some serious carroty (geo-plotting) issues... It looks like the maintainers of Cartopy are prioritizing getting a release out to fix this **very** soon.

One thing you could try right now to fix this is to specify 

```
conda install -c conda-forge matplotlib=3.5.2
```

or 

```
pip install matplotlib==3.5.2
```

srbrodzik, 2022-08-30:
Thanks everyone.  Will try the slightly older version of matplotlib (3.5.2)
and keeping check on the new cartopy release.

Stacy

srbrodzik, 2022-08-30:
OK, installed python3-matplotlib v3.5.2 and plots still don't show lat and
lon values on axes.  Here's what I have installed on my Debian 11 machine:

updraft:brodzik:99>dpkg -l | grep matplot
ii  python-matplotlib-data                        3.5.2-2
                                   all          Python based plotting
system (data package)
ii  python-mpltoolkits.basemap-data               1.2.2+dfsg-1
                              all          matplotlib toolkit to plot on
map projections (data package)
ii  python3-matplotlib                            3.5.2-2
                                   amd64        Python based plotting
system in a style similar to Matlab (Python 3)
ii  python3-mpltoolkits.basemap                   1.2.2+dfsg-1+b1
                           amd64        matplotlib toolkit to plot on map
projections (Python 3)
updraft:brodzik:100>dpkg -l | grep carto
ii  python-cartopy-data                           0.20.2+dfsg-2
                             all          cartographic library for Python
(package data)
ii  python3-cartopy                               0.20.2+dfsg-2
                             amd64        Cartographic library for Python 3
updraft:brodzik:101>dpkg -l | grep pyart
ii  python3-arm-pyart                             1.12.5-UW2
                                amd64        Py-ART: Python ARM Radar
Toolkit

The code runs fine on a Debian 10 machine running these packages.  However,
I plan to upgrade all our machines to Debian 11 sooner than later.

shear:brodzik:1113>dpkg -l | grep matplot
ii  python-matplotlib-data                        3.3.3-1
                                     all          Python based plotting
system (data package)
rc  python-matplotlib2-data                       2.2.3-6
                                     all          Python based plotting
system (data package)
ii  python-mpltoolkits.basemap-data               1.2.0+dfsg-1
                                all          matplotlib toolkit to plot on
map projections (data package)
ii  python3-matplotlib                            3.3.2-1
                                     amd64        Python based plotting
system in a style similar to Matlab (Python3)
ii  python3-mpltoolkits.basemap                   1.2.0+dfsg-1
                                amd64        matplotlib toolkit to plot on
map projections (Python 3)
shear:brodzik:1114>dpkg -l | grep carto
ii  python-cartopy-data                           0.19.0+dfsg-1~exp1
                          all          cartographic library for Python
(package data)
ii  python3-cartopy                               0.19.0+dfsg-1~exp1
                          amd64        Cartographic library for Python 3
shear:brodzik:1115>dpkg -l | grep pyart
ii  python3-arm-pyart                             1.11.2-1
                                    amd64        Py-ART: Python ARM Radar
Toolkit

mgrover1, 2022-09-01:
I will try to recreate this environment - thanks for sharing which packages you used.

srbrodzik, 2022-09-02:
Thanks for all your help with this.

Stacy

mgrover1, 2022-09-22:
@srbrodzik - I tracked down the issue. You will need to update your version of cartopy to use the new version of matplotlib

see this discussion https://github.com/SciTools/cartopy/pull/2054

srbrodzik, 2022-09-22:
> see this discussion SciTools/cartopy#2054

I'll take a look and let you know how it goes.

Thanks for the follow up.

Stacy

mgrover1, 2022-09-29:
@srbrodzik any updates here? Did this solve the issue?

srbrodzik, 2022-09-29:
I've asked our sys admin to install the updated packages but they're not
available yet.  I will resubmit the request.

I'll let you know what happens as soon as I can.  Thanks for following up
with me.

Stacy

srbrodzik, 2022-10-12:
Hey Max,

Sorry it took so long to get back.

The bad news is that the problem isn't fixed.  I've attached the two
versions of the image -- one using a Debian10 system and the other the
upgraded Debian11 system.

This is what is installed on the Debian10 system:

shear:~# apt list --installed | grep matplotlib
python-matplotlib-data/unknown,now 3.3.3-1 all [installed,automatic]
python-matplotlib/now 2.2.4-1 amd64 [installed,local]
python3-matplotlib/now 3.3.2-1 amd64 [installed,local]
shear:~# apt list --installed | grep cartopy
python-cartopy-data/unknown,now 0.19.0+dfsg-1~exp1 all
[installed,automatic]
python3-cartopy/unknown,now 0.19.0+dfsg-1~exp1 amd64 [installed]
shear:~# apt list --installed | grep pyart
python3-arm-pyart/unknown,now 1.11.2-1 amd64 [installed]

This is what is installed on the Debian11 system:

updraft:~# apt list --installed | grep matplotlib
python-matplotlib-data/now 3.6.0 all [installed,local]
python3-matplotlib/now 3.6.0 amd64 [installed,local]
updraft:~# apt list --installed | grep cartopy
python-cartopy-data/now 0.21.0+dfsg-2 all [installed,local]
python3-cartopy/now 0.21.0+dfsg-2 amd64 [installed,local]
updraft:~# apt list --installed | grep pyart
python3-arm-pyart/unknown,now 1.12.5-UW2 amd64 [installed]

I've also attached the code and the input file.
 cfrad.20220607_042450.162_to_20220607_043052.36...
<https://drive.google.com/file/d/1cA3Jow5HWj1k5CPw4kH0aANcSm2R4czn/view?usp=drive_web>

zssherman, 2024-05-14:
@srbrodzik Apologizes, late response, but just wanted to follow up, were you able to get the issue solved?

srbrodzik, 2024-05-14:
Hi Zach,

Honestly, this happened so long ago, I can't remember the details or if I
found a solution to the problem.

Go ahead and close this out.  If I have other issues down the line, I'll
contact you.

Stacy